drowe67/codec2-dev #9: CMake requires SVN

donmr, 2019-03-28:
In the top level CMakeLists.txt is code that gets the SVN version.  If you checkout from GIT into a new directory where SVN has not been used, AND it is named codec2-dev, then cmake gets an error:

CMake Error at /usr/share/cmake-3.10/Modules/FindSubversion.cmake:99 (message):
  Command "/usr/bin/svn info
  /home/don/Radio/Digital/FreeDV/Source/Git/codec2-dev" failed with output:

  svn: E155007: '/home/don/Radio/Digital/FreeDV/Source/Git/codec2-dev' is not
  a working copy

hobbes1069, 2019-03-28:
Yup, still some cleanup to do after the move. I'll take a look.

hobbes1069, 2019-03-28:
I'm not sure if it really does anything useful at this point but I have updated it to show the git short commit hash.

donmr, 2019-03-29:
Fixed

closed 2019-03-29 by donmr