drowe67/codec2-dev #14: Build Directory .travis.yml

srsampson, 2019-03-31:
The build directory may cause problems, as many octave tools were designed to look for "build_linux" and using the current "build" will fail.

Change Line 34
```
    - export BUILDDIR=${PWD}/build
```
To
```
    - export BUILDDIR=${PWD}/build_linux
```

db4ple, 2019-03-31:
The suggest change would not help. Travis builds linux in $BUILDSTD which is $BUILDDIR/normal, so the resulting build would be in build_linux/normal which is not better or worse than before:

https://github.com/drowe67/codec2/blob/4db67e8b0c6e0b15f7178237c979e14f0ebb9f6c/.travis.yml#L34

I was not aware of the hardcoded path into the build directory in some octave scripts. The INSTALL readme does not tell this either unfortunately:

https://github.com/drowe67/codec2/blob/4db67e8b0c6e0b15f7178237c979e14f0ebb9f6c/INSTALL#L22

Changing .travis.yml is no problem.

However, we should explicitly state the strongly recommended  nature of  "build_linux" directory for building linux, a similar thing seems to apply for the build_stm32 cmake location. And possible we should also let cmake check this.

This will help newcomers a lot, avoids a lot of frustration

srsampson, 2019-03-31:
I knew I would pick the wrong line to modify. I was looking for the easiest solution.

comments in "INSTALL" might be the quick solution.

Currently the travis builds are not having any problems.

Interestingly one of the cmake tests is "tofdm.m" which has a workaround builtin, which uses the PATH_TO_TOFDM assign. I don't think all the octave scripts have this kind of feature.

drowe67, 2019-04-25:
Has this one been resolved guys?

db4ple, 2019-04-25:
Long ago, travis uses build_linux and stm32/build_stm32 as advised...

closed 2019-04-25 by drowe67