drowe67/codec2-dev #20: STM32 Unittests Failing

donmr, 2019-04-11:
Many of the stm32 unittests are failing in the master branch, but not all.  I'll be digging into it tonight and tomorrow...

Don

drowe67, 2019-04-12:
Ach ... we need to stop breaking yr work Don!  Some tests perhaps.

db4ple, 2019-04-12:
Good Morning,
the culprit seem to be not so good settings in the link process, which I introduced to reduce the memory usage. Essentially we are missing support for %f in printf since I switched to the newlib "nano" libc which does not do float printf by default. 
I am not through all of the tests, but it looks good now once printf float gets enabled.

drowe67, 2019-04-12:
Fine Business @db4ple, BTW let me know if you want my Discovery/laptop turned on, I switched it off to save power ... happy to leave it it on if it's getting daily use.

db4ple, 2019-04-12:
Well, I will work on the "remoting" of stm32 unittests locally first in my internal network. I'll let you know once we are good to integrate this into travis and need to test it with your laptop.

@donmr : BTW, even now not all test pass (I am currently running the ofdm tests) but this seems to be due to slight changes in the output. E.g. the test_ofdm_demod_AWGN now reports sllightly different (lower) BER which lets the test report a FAIL.  I think/hope is an "acceptable" fail and can be resolved by changing the reference result. But this is not for me to decide/fix...

drowe67, 2019-04-12:
> @donmr : BTW, even now not all test pass (I am currently running the ofdm tests) but this seems to be due to slight changes in the output. E.g. the test_ofdm_demod_AWGN now reports sllightly [...]

The measured BER test is a random variable, whose mean is the actual BER.  Every time we run a test, we sample that random variable, and get a slightly different result. Someone who knows something about statistics (not me unfortunately!) could work out a 5% confidence interval or upper bound.  What we are trying to trap with these tests is an implementation error that has pushed the mean BER up.

donmr, 2019-04-12:
Is the fix going to be commited??
Yes, anyone changing the STM32 build files needs to run the tests we have.
I don't know when this change came in, I pulled the proposed changes I saw and it all worked for me.
Somehow this got missed, that happens...

donmr, 2019-04-12:
If you tell me what you changed to get it mostly working, I'll put that in and work on the rest of the issues.
It looks like the "specs" were changed to nano and then some magic puts rdimon back in.  Just changing back to rdimon won't work.

db4ple, 2019-04-12:
Hi @donmr, see #21 . It is just `-u_printf_float` after `--specs=nano.specs`

db4ple, 2019-04-12:
#21 fixed all test but the tst_ofdm_demod_AWGN / tst_ofdm_demod_fade and tst_ofdm_demod_lpcd_AWGN / tst_ofdm_demod_lpcd_fade (which showed the mentioned differences)

donmr, 2019-04-12:
I merged this with a new branch made from master and all of the ofdm tests passed.  Once it is in master we should both run the tests again and post any errors.

drowe67, 2019-04-12:
Good work guys.

@donmr can you pls [document](https://github.com/drowe67/codec2/blob/master/stm32/unittest/README_unittest.txt) (or even better automate) the procedure to run **all** stm32 tests?  I'd like to repeat it here, in particular before I merge https://github.com/drowe67/codec2/pull/8

donmr, 2019-04-12:
Created PR #22 with instructions added and updated

drowe67, 2019-07-05:
Can we close this issue now?  All stm32 tests are passing (even on Ubuntu 18 machines).

donmr, 2019-07-05:
Yes

closed 2019-07-05 by donmr