drowe67/codec2-dev #23: STM32 remote testing with OpenOCD

db4ple, 2019-04-13:
@donmr , @drowe67 
I converted the existing run_stm32_prog script, so that is uses openocd via ssh.
Some but not all tests passed as expected. When run with semihosting through gdb (i.e. remote fileio) some tests produce to short or too large output. When run exactly the same way using openocd BUT using local semihosting fileio (openocd command: semihosting_fileio disable vs. semishosting_fileio enable), everything passes as expected. Which means, there is an issue related to the fileio stuff either in openocd or in the gdb or in the actual binary. Further drilling down let me find that fread behaves sometimes strange (reads too much or not enough). There are two workarounds to fix that: 

1. We use `setbuf(fin,NULL);`  to disable buffering of the input file. Works but is ultra slow (6min runtime for a single tests vs. ~30s ).
1. We switch to basic fileio using `open/read/write` (as opposed to `fopen/fread/fwrite). This is almost as fast as the original code and does not seem to have the problems. 

Or we leave it as is and find another way to remotely execute the tests. But honestly, I think the conversion (option 2) is feasible. Takes a couple of minutes per test and is basically a no-brainer.

donmr, 2019-04-13:
I have tried to use #2 for newer tests.  There is no need for additional 
buffering by stdio.

We should also check that all tests close the files to flush them.

db4ple, 2019-04-13:
Thanks!
Using the "open/read/write/close" has the benefit of not requiring explicit flush before close.
I will run all existing tests using openocd and convert those failing because of fread in a first round. 
Second round will include all remaining tests for consistency.

db4ple, 2019-04-13:
@donmr : I noticed that the tst_codec2_enc_1600 test does not pass anymore. Reason is simple:
The new CMakeLists.txt disables mode 1600 (only 1300 and 700C are enabled). Reason for that is that FreeDV 1600 actually uses codec2 1300 not codec2 1600.
So we have a couple of options again:

1. Change the tests to use the codec2 1300 (which is in fact the most popular one and the one used in sm1000). Requires some minor edits to the tests.
2. Change the compile settings to include 1600 as well. This would increase the footprint of binaries and test a mode not widely used.

This time I vote for 1.

donmr, 2019-04-13:
My mistake.  Since the SM1000 use  FDV 1600 I made that test to verify my 
test code.  I can change it to  1300 instead.

Don

db4ple, 2019-04-13:
@donmr : In my referenced commit  https://github.com/db4ple/codec2/commit/7832e861808c0dd664a9a9507db93f419910e474 I just did that and a couple of other things. 
Once the travis passes, I will create a pull request from it (hoping for your approval or comment)

db4ple, 2019-04-25:
Done with this (it works)

closed 2019-04-25 by db4ple