drowe67/codec2-dev #26: Discuss removal / replacement of outdate stuff from codec2

db4ple, 2019-04-19:
We will use this comment to consolidate the discussion, so that you don't have to go through all comments if this gets a longer discussion:

To be deleted:
1. stm32/Makefile, stm32/unittest/src/Makefile -> We are using cmake now.

drowe67, 2019-04-25:
Perhaps we could also use this as a place to discuss what's required to get the fine work @donmr running on the SM1000.  We must be very close :-)

donmr, 2019-04-25:
Yes, once we stop having issues with the tests...

I think the main thing next for me is to build a 700D only version and
see how that does.  It will compile but can it malloc enough??

Adding the ability to select a mode can come later.  I was tring to do
this before but it adds too much complexity and might need more memory.

I'll be busy with ARES and SAR exercises this weekend but next week that
is on my plan.

Don

drowe67, 2019-04-26:
Re stm32 test issues, there are a couple of outcomes to the work in https://github.com/drowe67/codec2/pull/25:

1. Fix the differences between Ubuntu 16 and 18 to ground, no matter how long it takes.
1. Adopt statistical, rather than bit exact/vector comparison conditions for a pass on the tests in question, so they pass on Ubuntu 16 and 18.
1. Given we agree there is no problem with the stm32 code, and considering the time we have spent on it, put the problem aside for now and move on.  We can agree to run stm32 tests on Ubuntu 16 before any merge.

db4ple, 2019-04-28:
Hi,
the difference is also between Fedora 24 and 25 (glibc 2.23 / 2.24). Fedora 24 passes, 25 does not.
I am currently verifying this with Ubuntu and try to see when it starts to fail on bear.

drowe67, 2019-05-02:
Remove a some unused modes in Codec 2 and FreeDV:

In Codec 2:

#define CODEC2_MODE_3200 	0
#define CODEC2_MODE_2400 	1
#define CODEC2_MODE_1600 	2
~~#define CODEC2_MODE_1400 	3~~
#define CODEC2_MODE_1300 	4
#define CODEC2_MODE_1200 	5
~~#define CODEC2_MODE_700  	6~~
~~#define CODEC2_MODE_700B 	7~~
#define CODEC2_MODE_700C 	8
#define CODEC2_MODE_450 	10
#define CODEC2_MODE_450PWB 	11

In FreeDV:

#define FREEDV_MODE_1600        0
~~#define FREEDV_MODE_700         1~~
~~#define FREEDV_MODE_700B        2~~
#define FREEDV_MODE_2400A       3
#define FREEDV_MODE_2400B       4
#define FREEDV_MODE_800XA       5
#define FREEDV_MODE_700C        6
#define FREEDV_MODE_700D        7
#define FREEDV_MODE_2020        8

Happy for anyone to raise a PR and work on that.  Pls don't renumber the mode ID number, as that might break ABI compatibility for some users.

srsampson, 2019-05-03:
I was thinking, that if the mode structures were stored in flash, that different modes could be enabled by first loading the correct structure config, and then initializing. Same with transmit and receive on half-duplex. When you push PTT it loads the TX structure config, and when you release it loads the RX structure config. Maybe an alarm when the user pushes the PTT faster than once per second (ignore it then). Thus RAM is re-used, rather than depleted by multiple modes.

drowe67, 2019-07-17:
@db4ple - is this Issue still relevant?

db4ple, 2019-07-17:
It was meant to be a place to collect ideas what to remove and to capture this kind of discussion in a single visible "thread". However, if we don't (want to) use it like this, it should be closed.

drowe67, 2019-07-17:
Cool, I'm OK to leave it open.

kantooon, 2019-10-25:
Please don't remove Codec2 mode 1400. I am using it: https://github.com/qradiolink/qradiolink/blob/1d38b4366fcb9741a143f393498157f96c3e0685/audio/audioencoder.cpp#L44

drowe67, 2019-11-29:
Removed Codec 2 700/700B and FreeDV 700/700B in https://github.com/drowe67/codec2/pull/90.

If there is anything else that anyone feels we need to be remove pls open with a specific issue around that chunk of code/feature.

closed 2019-11-29 by drowe67