drowe67/codec2-dev #29: 700D failing to get interleaver sync

drowe67, 2019-04-23:
I have an off air FreeDV 1.3.1 700D sample that won't decode using codec2/master & freedv-gui/master built from Git Head.  It gets modem sync but not interleaver sync.

Suspect this is related to the work @donmr and myself did around low memory 700D optimisations for the stm32 in freedv_api.c, I have done some work on this in the codec2/brad-2020 and codec2/dr-2020 branches but it's still not decoding successfully.

We'll need to fix this before another FreeDV release.  A good start would be back tracking through Git revisions and perhaps testing with freedv_rx.

db4ple, 2019-04-23:
David, I guess you know `git bisect`. For those who don't: It helps you to find the git commit which broke something quickly. You need to give it one known to  work commit (ideally not too old) and will checkout commits and ask if they are good or bad. Often needs only a few steps. Sometimes fails.

drowe67, 2019-04-23:
Thanks Danilo.  Another idea is to cross check and earlier freedv_tx with a current freedv_rx to confirm the bug exists at the API level.  I'll think about it some more.

drowe67, 2019-04-24:
OK the culprit is me in 20d24ed2b, we changed the tuning of the 700D modem by one carrier I would guess

drowe67, 2019-04-25:
All fixed now

closed 2019-04-25 by drowe67