drowe67/codec2-dev #36: CMake and Mode Flags

donmr, 2019-05-07:
in stm32/CMakeLists.txt there are some "add_definitions" entries.  I believe that those apply to all targets.  I want to create separate SM1000 binaries for 1600 and 700D (merging may come later when the each work separately).  I can imagine may future cases like this.

I'm not very familiar with CMake.  It looks like there are some new options that support this but that version isn't on my system and probably isn't on many.

Can a CMake "expert" help?

drowe67, 2019-05-07:
@hobbes1069 is our cmake guy :-)

However perhaps you just need two versions of ```sm1000_main.c```, the current one that kicks off 1600, and a new one that starts 700D.  CMakeLists.txt could be set up to build them both, it's really just a copy/past edit of the current sm1000 target.

donmr, 2019-05-07:
Yes there will be 2 mains But there is one CMakeLists.txt which controls 
all of the targes.

Don

drowe67, 2019-05-07:
Sure - well @hobbes1069 or myself can help you with the CMakeLists.txt.  Just ping us when you are ready for that stage.

donmr, 2019-05-07:
That is what I though I was doing with this issue, pinging whoever knew how to do it.

hobbes1069, 2019-05-08:
I can update the cmake config but I need to understand what needs to be target specific. All that is required is to change from a global "add_definitions" to a target specific "target_compile_definitions"

https://cmake.org/cmake/help/latest/command/target_compile_definitions.html

donmr, 2019-05-08:
There is a branch donmr_sm1000_700D.  I checked in a version of stm32/CMakeLists.txt which has 2 targets, sm1000 and sm1000_700d.

sm1000 should just set FREEDV_MODE_1600_EN and CODEC2_MODE_1300_EN.
sm1000_700d shoud just set FREEDV_MODE_700D_EN and CODEC2_MODE_700C_EN.

db4ple, 2019-05-08:
@donmr 
I push a commit to build sm1000 and sm1000_700D. Instead of using sm1000_main.c and sm1000_700D.c I added the necessary code detect which mode is available to the sm1000_main.c.

closed 2019-05-08 by donmr