drowe67/codec2-dev #48: STM32 tests and library path

donmr, 2019-07-02:
We need to be sure tests are using the correct codec2 shared library to create input and reference files.  A while ago I changed stm32/unittest/scripts/setup.sh to put the "current" library directory at the front of LD_LIBRARY_PATH.

Today I tried using the CMake version of the stm32 tests.  I see the same error where it is finding a different library.  I don't know how this path is set in CMake but this needs to be fixed.

drowe67, 2019-07-03:
Thanks @donmr.  Yes I think the x86 Ctests automatically use a path local to the build, we may need a similar system in the stm32 tests.

hobbes1069, 2019-07-05:
Did you guys get this sorted? I haven't had a lot of time but i have been reading everything. Let me know if you need help.

drowe67, 2019-07-05:
I think this issue is still open @hobbes1069 .  We run some x86 codec 2 programs in codec2/stm32/unittest/scripts to generate test data for the stm32 tests.  @donmr has found that these can clash with other versions of codec 2 on the same machine.

Possible solutions are:
1. Ctest configuration such that the PATH and LD_LIBRARY_PATH are set.
1. Modifying the scripts to use a local version of the codec 2 executables.

A stm32 Discovery card is needed to run the stm32 tests.

donmr, 2019-07-05:
When I run manually I use stm32/untests/scripts/setup.sh to set LD_LIBRARY_PATH and PATH.
I think this should be done for CMake too but I don't know how that happens in CMake.

I don't like to add this to run_stm32_tst as the user should be able to control the environment outside the script.  But it could be added there if it is not simple inside CMake.

drowe67, 2019-07-06:
Another approach is to add the full path to codec2/build_linux every time we use an x86 codec 2 executable (so modify all the _setup and _check scripts).  Not sure if it  will pick up the correct .so from that dir too.

I don't have a strong preference on how we do it.

closed 2019-12-01 by drowe67