drowe67/codec2-dev #56: Function depends on double

srsampson, 2019-07-24:
Ref:
https://github.com/drowe67/codec2/blob/master/src/cohpsk.c#L1114

In stm32 compile there are many float to double warnings. If I change the double to float and the floor/ceil to floorf/ceilf, it fails the tcohpsk unittest with an assert on nout.

Which leads me to think that the stm32 version probably wouldn't work, even though it compiles.

I'm wondering if we can find the reason why doubles are needed and fix it so floats will work.

I don't think this program is used in stm32 right now, so probably not an immediate concern.

drowe67, 2019-07-24:
Hi @srsampson yes I'm sure it can be fixed, so that the warnings go away and the test passes. I recall we had a similar problem with the ofdm modem, when setting the carrier frequencies.

As a start, perhaps we could raise e PR and fix all the warnings that don't affect the test, that would at least minimise the problem.

I've found it's a balancing act to fix warnings from the stm32 and x86 gcc compilers at the same time.

srsampson, 2019-07-24:
I made a debug printf() to see the error. It sort of becomes off-by-one.

```
      t1 = floor(tin);           // double
      t1a = floorf(tina);      // float
      t2 = ceil(tin);            // double
      t2a = ceilf(tina);       // float
      printf("t1: %d t1a: %d t2: %d t2a: %d\n", t1, t1a, t2, t2a);

t1: 10200 t1a: 10200 t2: 10201 t2a: 10201
t1: 10201 t1a: 10201 t2: 10202 t2a: 10202
t1: 10202 t1a: 10202 t2: 10203 t2a: 10203
t1: 10203 t1a: 10202 t2: 10204 t2a: 10203 <---- oops
t1: 10204 t1a: 10203 t2: 10205 t2a: 10204
t1: 10205 t1a: 10204 t2: 10206 t2a: 10205
```

drowe67, 2019-07-25:
Hmm, so maybe it just happens to round to an integer so ```floor(integer)== ceilf(integer)``` maybe we could redefine ```t2 = t1+1```

I forget what that code is, can you point me at the source pls?

srsampson, 2019-07-25:
I did a pull request (ss-fs-offset) just to get rid of the warnings. But maybe highlight the code that might need a float version if this needs to run in firmware.

srsampson, 2019-07-25:
Oh, I don't think that will work, as the differences become even more spread out than just 1.

closed 2019-07-25 by srsampson