drowe67/codec2-dev #60: Add a release?

FFY00, 2019-07-30:
Could you please add a release so that this can be packaged properly? Thank you

drowe67, 2019-08-03:
We're kind of in development stage atm, in particular with FreeDV 2020

FFY00, 2019-08-04:
I understand but there are projects using this in production. Maybe you could add a pre-release? I just want to have a commit marked as stable.

drowe67, 2019-08-04:
OK.  I'm new to Git/GitHub.  How would you like me to mark the commit?  Just a comment in the comment message?

FFY00, 2019-08-04:
Create a tag, you can do it directly in the web interface by going to the releases section 😁

hobbes1069, 2019-08-07:
I can add a tag/release, but I'm unsure what to tag it as :) What feature is in master that's missing in the previous release?

drowe67, 2019-08-07:
Good question!  There has been quite a lot of work:
+ FreeDV 2020 mode (beta at this stage)
+ 700D ported to the SM1000, new SM1000 build system, unit test system, v2 firmware release
+ OFDM modem is now reconfigurable and has had acquisition improvements
+ FreeDV 700C freq offset bug fix
+ Codec 2 700C equaliser (beta)

FFY00, 2019-08-07:
> I can add a tag/release, but I'm unsure what to tag it as :) What feature is in master that's missing in the previous release?

API changes. I am particularly interested is getting a release that works with sdrangel.

drowe67, 2019-08-07:
> > I can add a tag/release, but I'm unsure what to tag it as :) What feature is in master that's missing in the previous release?
> 
> API changes. I am particularly interested is getting a release that works with sdrangel.

Can you pls be more specific about what API changes are causing issues?  There haven't been many API changes recently, just a few more mode options.

DG9BFC, 2019-08-16:
> Good question! There has been quite a lot of work:
>
>     * FreeDV 2020 mode (beta at this stage)
>
>     * 700D ported to the SM1000, new SM1000 build system, unit test system, v2 firmware release [...]

when do we have a new windows build to test ??? i guess that equalizer can really improve things (tested freedv on qo100 ... wow that works so superb

about porting 700d to stm32 ... maybe we can also add it in mchf firmware (wink wink)
greetz sigi dg9bfc

db4ple, 2019-08-16:
@DG9BFC : 700d is already in the UHSDR firmware but only for STM32F7/H7 TRX. Most likely 700d will not work on the STM32F4 mcHF TRX because the SDR IQ signal processing requires too much time and there is simply not enough processing time (and memory) left to decode / encode 700d in realtime. 

The SM1000 does only the audio signal processing and because of that and the work of the Don and others is now able to run 700D on the STM32F4. I am not saying it is definitely impossible on the F4 mcHF but if possible at all, it is a very big challenge to squeeze all of this into a little STM32F4.

drowe67, 2019-08-16:
I'm placing development builds for windows [here](http://rowetel.com/downloads/freedv/)

DG9BFC, 2019-08-20:
hi david .... i know your devel folder :-) (and tested newest on qo100)
cant wait to have all those new functions working
on the sat we have a very stable path and freedv works superb
sadly 2020 mode not included yet ... but i guess any mode that works on shortwave ... should work even better on that 72000km path :-)

DG9BFC, 2019-08-20:
just a short report over a few tests i made on qo100 ...
for a good readable ssb signal i need to use 3 to 5 watt (readable lower but i wrote GOOD readable)
some modes can be decoded well if i use just half a watt or so ... so i can say over that 72000km path freedv works much better as analog ssb ... go figure
with half a watt to 1 watt used ... ssb us just even redable (but only for a trained ear and with adding software noise reduction as a bit of help)
with 1w in freedv used ... crystal clear audio with absolutely no dropouts or noise
WOW THAT ROCKS :-)

drowe67, 2019-08-20:
Thanks for that report.  A Geo satellite channel is much nicer than HF, it's almost pure AWGN with no fading, so modes like 700D will work 4dB better than HF for a given SNR.

I think all FreeDV 1.4 development versions include 2020.  If you have the button greyed out it's because your computer does not support AVX.  Please the [FreeDV User Manual](https://github.com/drowe67/freedv-gui/blob/master/USER_MANUAL.md).  if you can't see the 2020 mode button pls email me a screen shot and the file name of the 1.4 version you are using.

hobbes1069, 2019-08-20:
Did you guys figure things out or do I still need to tag a release? I brought back a terrible stomach bug from China with me and I'm just now starting to feel better.

DG9BFC, 2019-08-20:
greyed out ...
hmmm i have a quad core atom tablet
it is a modern cpu ... not very fast but a quadcore .... 4 gb ram running w10
i could run coreinfo ... but i can not read fast enough (runs in a box and closes within a second)
so coreinfo exe did not help

drowe67, 2019-08-22:
Sigi - tried to reply to yr email but am being bounced:

AVX is hardware acceleration for the LPCNet speech codec used in the new 2020 mode.  Sorry, it won't run without AVX.

BTW Gerhard OE3GBB has run 2020 over that satellite.  We did some work to modify the modem phase estimator for that channel - see user manual.

Probably best to move this discussion on to the codec 2 mailing list.

Thanks,
David

DG9BFC, 2019-08-22:
hmmm ... i though it would be possible to compile a version that runs without avx ... so ... seems as i have no chance to run twenty twenty on my tablet :-(
if that avx function is needed for that codec used ... and the codec used is needed to run that mode ... then there is no chance for me ... BUHUUUU that is so sad
lower speeds run on stm32 and you have to use a modern fast pc for that new mode?? hmmmm .... ok so be it ...
you can write to dg9bfc(at) freenet dot de

doronbehar, 2020-10-09:
Hey all, came around this old issue. On https://github.com/qradiolink/qradiolink/issues/67 we realized that many packagers are having trouble building gnuradio with freedv support because codec2's freedv support is not present on most distros, because there is no release of codec2 that includes it.

Could you please publish a release so we packagers could enjoy these cool features properly? :pray:

drowe67, 2020-10-09:
Hi @doronbehar sure we can look into this. @hobbes1069 is this something you could handle please?  I'm not quite sure how to do this myself.

hobbes1069, 2020-10-12:
Sure, just latest master?

doronbehar, 2020-10-12:
It turns out we've had an issue with GR3.8 not detecting codec2 v0.9.2 properly:

```
gnuradio> CMake Error at gr-vocoder/lib/CMakeLists.txt:23 (add_library):
gnuradio> Target "gnuradio-vocoder" links to target "CODEC2::CODEC2" but the target
gnuradio> was not found.  Perhaps a find_package() call is missing for an IMPORTED
gnuradio> target, or an ALIAS target is missing?
gnuradio> CMake Error at gr-vocoder/lib/CMakeLists.txt:23 (add_library):
gnuradio> Target "gnuradio-vocoder" links to target "GSM::GSM" but the target was not
gnuradio> found.  Perhaps a find_package() call is missing for an IMPORTED target, or
gnuradio> an ALIAS target is missing?
gnuradio> CMake Error at cmake/Modules/UseSWIG.cmake:573 (add_library):
gnuradio> Target "vocoder_swig" links to target "CODEC2::CODEC2" but the target was
gnuradio> not found.  Perhaps a find_package() call is missing for an IMPORTED
gnuradio> target, or an ALIAS target is missing?
gnuradio> Call Stack (most recent call first):
gnuradio> cmake/Modules/GrSwig.cmake:137 (swig_add_library)
gnuradio> gr-vocoder/swig/CMakeLists.txt:47 (GR_SWIG_MAKE)
gnuradio> CMake Error at cmake/Modules/UseSWIG.cmake:573 (add_library):
gnuradio> Target "vocoder_swig" links to target "GSM::GSM" but the target was not
gnuradio> found.  Perhaps a find_package() call is missing for an IMPORTED target, or
gnuradio> an ALIAS target is missing?
gnuradio> Call Stack (most recent call first):
gnuradio> cmake/Modules/GrSwig.cmake:137 (swig_add_library)
gnuradio> gr-vocoder/swig/CMakeLists.txt:47 (GR_SWIG_MAKE)
gnuradio> CMake Error at gr-vocoder/lib/CMakeLists.txt:23 (add_library):
gnuradio> Target "gnuradio-vocoder" links to target "CODEC2::CODEC2" but the target
gnuradio> was not found.  Perhaps a find_package() call is missing for an IMPORTED
gnuradio> target, or an ALIAS target is missing?
```

And previously:

```
gnuradio> CMake Warning (dev) at /nix/store/70y1dj79fq7f6486y8clrgngrcfr303q-cmake-3.18.2/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
gnuradio> The package name passed to `find_package_handle_standard_args` (PkgConfig)
gnuradio> does not match the name of the calling package (Codec2).  This can lead to
gnuradio> problems in calling code that expects `find_package` result variables
gnuradio> (e.g., `_FOUND`) to follow a certain pattern.
gnuradio> Call Stack (most recent call first):
gnuradio> /nix/store/70y1dj79fq7f6486y8clrgngrcfr303q-cmake-3.18.2/share/cmake-3.18/Modules/FindPkgConfig.cmake:59 (find_package_handle_standard_args)
gnuradio> cmake/Modules/FindCodec2.cmake:24 (INCLUDE)
gnuradio> gr-vocoder/CMakeLists.txt:48 (find_package)
gnuradio> This warning is for project developers.  Use -Wno-dev to suppress it.
gnuradio> CMake Warning (dev) at /nix/store/70y1dj79fq7f6486y8clrgngrcfr303q-cmake-3.18.2/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
gnuradio> The package name passed to `find_package_handle_standard_args` (LIBCODEC2)
gnuradio> does not match the name of the calling package (Codec2).  This can lead to
gnuradio> problems in calling code that expects `find_package` result variables
gnuradio> (e.g., `_FOUND`) to follow a certain pattern.
gnuradio> Call Stack (most recent call first):
gnuradio> cmake/Modules/FindCodec2.cmake:78 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
gnuradio> gr-vocoder/CMakeLists.txt:48 (find_package)
gnuradio> This warning is for project developers.  Use -Wno-dev to suppress it.
gnuradio> -- Could NOT find LIBCODEC2 (missing: LIBCODEC2_INCLUDE_DIRS)
gnuradio> CMake Warning (dev) at /nix/store/70y1dj79fq7f6486y8clrgngrcfr303q-cmake-3.18.2/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
gnuradio> The package name passed to `find_package_handle_standard_args` (PkgConfig)
gnuradio> does not match the name of the calling package (GSM).  This can lead to
gnuradio> problems in calling code that expects `find_package` result variables
gnuradio> (e.g., `_FOUND`) to follow a certain pattern.
gnuradio> Call Stack (most recent call first):
gnuradio> /nix/store/70y1dj79fq7f6486y8clrgngrcfr303q-cmake-3.18.2/share/cmake-3.18/Modules/FindPkgConfig.cmake:59 (find_package_handle_standard_args)
gnuradio> cmake/Modules/FindGSM.cmake:24 (INCLUDE)
gnuradio> gr-vocoder/CMakeLists.txt:64 (find_package)
gnuradio> This warning is for project developers.  Use -Wno-dev to suppress it.
gnuradio> CMake Warning (dev) at /nix/store/70y1dj79fq7f6486y8clrgngrcfr303q-cmake-3.18.2/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
gnuradio> The package name passed to `find_package_handle_standard_args` (LIBGSM)
gnuradio> does not match the name of the calling package (GSM).  This can lead to
gnuradio> problems in calling code that expects `find_package` result variables
gnuradio> (e.g., `_FOUND`) to follow a certain pattern.
gnuradio> Call Stack (most recent call first):
[... 2 lines cut ...]
gnuradio> This warning is for project developers.  Use -Wno-dev to suppress it.
gnuradio> -- Could NOT find LIBGSM (missing: LIBGSM_INCLUDE_DIRS)
gnuradio> CMake Warning (dev) at /nix/store/70y1dj79fq7f6486y8clrgngrcfr303q-cmake-3.18.2/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
gnuradio> The package name passed to `find_package_handle_standard_args` (PkgConfig)
gnuradio> does not match the name of the calling package (GSL).  This can lead to
gnuradio> problems in calling code that expects `find_package` result variables
[... cut ...]
      "-DLIBCODEC2_HAS_FREEDV_API=ON"
      "-DLIBGSM_LIBRARIES=${gsm}/lib/libgsm.so"
      "-DLIBGSM_INCLUDE_DIRS=${gsm}/include/gsm"
```

Where `${gsm}` and `${codec2}` are "installed" - it's a Nix thing to use this syntax. I'm not sure whose fault is this, but I'd be happy to tell you how do we get along with any new release, and whether these cmake flags are needed. I'm reluctant to say it's GR3.8 not doing a good job detecting codec2 properly, since it's gsm that's also not detected properly without these flags.

hobbes1069, 2020-10-12:
Well I don't need any special definitions to get FreeDV to work on Fedora so I'm wondering what's up. I would check on the import files and see if they're being created correctly. On Fedora I have:

```
$ cat ./usr/lib64/cmake/codec2/codec2-config-relwithdebinfo.cmake
#----------------------------------------------------------------
# Generated CMake target import file for configuration "RelWithDebInfo".
#----------------------------------------------------------------

# Commands may need to know the format version.
set(CMAKE_IMPORT_FILE_VERSION 1)

# Import target "codec2" for configuration "RelWithDebInfo"
set_property(TARGET codec2 APPEND PROPERTY IMPORTED_CONFIGURATIONS RELWITHDEBINFO)
set_target_properties(codec2 PROPERTIES
  IMPORTED_LOCATION_RELWITHDEBINFO "${_IMPORT_PREFIX}/lib64/libcodec2.so.0.9"
  IMPORTED_SONAME_RELWITHDEBINFO "libcodec2.so.0.9"
  )

list(APPEND _IMPORT_CHECK_TARGETS codec2 )
list(APPEND _IMPORT_CHECK_FILES_FOR_codec2 "${_IMPORT_PREFIX}/lib64/libcodec2.so.0.9" )

# Commands beyond this point should not need to know the version.
set(CMAKE_IMPORT_FILE_VERSION)
```

doronbehar, 2021-02-23:
The situation with GR3.9 detection of codec2 is even worse: https://github.com/gnuradio/gnuradio/issues/4278