drowe67/codec2-dev #65: Horus LDPC LLR

drowe67, 2019-08-10:
Issue with LLR mapping of rthe LDPC code used in Horus  SSTV applications. See https://github.com/drowe67/codec2/pull/61 for work required.

ED6E0F17, 2020-06-02:
I added a scaling feature to "ldpc_noise" so that you can instantly see that scaling up or down by a factor of 10 seriously degrades performance: e322be7b13148953b563d585326bba8685d84e42 , and that adding normalisation resolves the issue without cost. You can check each of the codes that you use, and integrate this into your normal unit tests if that would be helpful.

Test with: 
    ./ldpc_enc /dev/zero - --sd --code HRA_112_112 --testframes 100|./ldpc_noise - - 0 0.1 2>/dev/null | ./ldpc_dec - /dev/null --code HRA_112_112 --sd --testframes 2>tmp ; grep BER tmp

drowe67, 2020-06-02:
@ED6E0F17 thanks.  Maintenance for Horus work is being moved out of codec2 into https://github.com/drowe67/hab, which @darksidelemm will be maintaining.

closed 2020-06-02 by drowe67

darksidelemm, 2020-06-02:
Sounds like this is more related to the LDPC code generally, not just the particular code used by horus?

darksidelemm, 2020-06-02:
I did have some questions about the SD output from the FSK modem - it looks to go far above and below 0, presumably related to SNR. Is this the expected behaviour?
I guess this would make sense if those values (scaling with SNR) were log'd - as this would be a LLR almost by itself? Not sure about the theory on this one.

drowe67, 2020-06-03:
This issue is specifically for the 2FSK mapping, used only by Wenet.  There are other functions for mapping QPSK symbols to LLRs.

It's been about 4 years since I worked with this, so the details are fuzzy for me.  I didn't touch the FSK soft decision code in the recent refactoring so I presume the current code worked with Wenet.  I imagine there is (or rather was, as @ED6E0F17 pointed out when he raised this issue some time ago) a normalising step in the LLR calculations.

As Bill pointed out the FSK to LLR mapping is not straight forward, so I guess we had some sort of approximation that (clearly) worked well for Wenet.

The CML library also has a SD->LLR function for a variety of modulation methods, I recently used it for QAM16, so it might have a 2FSK version.