drowe67/codec2-dev #102: problem with ofdm.c

dave-g8kbb, 2020-04-28:
Apologies if this is the wrong place to raise this.
I think there nay be a problem with ofdm.c. James Ahlstom's quisk opens two receiver channels by calling freedv_open() twice. Quisk currently uses version 12, and this works fine. In version 13 however, ofdm.c introduced some static variables, for example tx_uw_syms. When quisk closes the first receiver channel, ofdm_destroy() calls FREE() to free up the memory allocated. When the second channel is closed, the call to ofdm_destroy() causes a memory exception which crashes quisk. Quisk is a python application that calls codec2 as a DLL via a python extension dll.
Should these variables not be part of the OFDM structure?
Dave

drowe67, 2020-04-28:
@dave-g8kbb thanks for your report.  Yes I can see how problem might crop up if there are two instances created of the modem, something we've not encountered until now.

@srsampson what do you think?

I'm happy to review a PR on fixing this one if either of you want to work on it.

dave-g8kbb, 2020-04-28:
Hi @drowe67 I'm happy to help in any way I can. I'm no expert on the internals of codec2 though. Also first time I've raised a query like this. What would you like me to do?
Best regards
Dave

srsampson, 2020-04-28:
The theory is, that with the SM1000 we don't want to allocate the filter memory as it is too large. So we moved the allocate out of ofdm_create(). The reference in ofdm_destroy() is an error then. freedv_api should handle the calls. I removed it from ofdm_destroy().

drowe67, 2020-04-28:
@srsampson is working on this on in https://github.com/drowe67/codec2/pull/103, will continue conversation there.

srsampson, 2020-04-29:
I created a new pull and just moved the unique word allocations as discussed.

Not sure if that resolves anything, but it does pass compile and tests.

drowe67, 2020-04-29:
Nice work @srsampson :+1: 

@dave-g8kbb would you be able to test Steve's branch in your application please?

dave-g8kbb, 2020-04-29:
@drowe67 I downloaded the branch, and it rebuilt without errors. I patched into quisk and it no longer crashes when switching modes.
@srsampson many thanks for the prompt work.
I'll carry on today retesting the complete system including some on-air tests - will let you know how it goes.

Best regards and once again many thanks.
Dave

srsampson, 2020-04-29:
Thanks Dave, hope all works fine!

Steve

dave-g8kbb, 2020-04-29:
Seems to work - I have changes to make to quisk now for 2020 and 2400B as the code presumes 8000 samples per second everywhere but so far so good. 1600 / 700C / 700D / 800XA fine opening 2 channels concurrently.

srsampson, 2020-04-29:
Static variables are kind of old school, from the dark ages before threading. But we can weed them out as developers use the code! I kind of misread the problem on the first go, but we finally got there.

drowe67, 2020-04-29:
Thanks guys - great work :1st_place_medal:

closed 2020-04-29 by drowe67