drowe67/codec2-dev #105: Possible error - uninitialised variables

dave-g8kbb, 2020-05-02:
Hello

Think there may be a problem with uninitialised variables when running 800XA mode.
Running under windows 10, codec2 was crashing. Tracing it, it was caused by msvcrt!memmove called from freedv_data_channel_rx_frame() in freedv_data_channel.c. It appears that deframer->fdc->packet_rx_cnt and packet_tx_cnt are not correctly initialised. They are created by malloc() not calloc() in freedv_data_channel_create() but unlike packet_tx_size do not appear to be initialised. This means that in freedv_data_channel_rx_frame() at line 196 the memcpy(fdc->packet_rx + fdc->packet_rx_cnt, data, copy_bits); carries a random (large) value for the memcpy location in the event that one of the previous conditionals are not taken (which sometimes happens - this crash was a little random).

The effect of this may I suppose be system / compiler dependent but for me it ends up causing memory segmentation errors as the values appear random.
The build system is msys2 64 bit / mingw64 with gcc version 9.3.0. In regard of flags, the cmake command I used was:
cmake -G "MSYS Makefiles" -DLPCNET_BUILD_DIR=~/LPCNet/build_win64 -DCMAKE_BUILD_TYPE=Debug -DUNITTEST=OFF -DCMAKE_SYSTEM_PROCESSOR=x86_64 ../

Adding this to freedv_data_channel_create() seems to cure the problem:
    fdc->packet_rx_cnt = 0;
    fdc->packet_tx_cnt = 0;

Best regards
Dave

srsampson, 2020-05-02:
I'd just replace it with the following, as packet_tx_size is also left dangling:
```
struct freedv_data_channel *freedv_data_channel_create(void)
{
    struct freedv_data_channel *fdc = (struct freedv_data_channel *) calloc(1, sizeof(struct freedv_data_channel));
    if (fdc == NULL)
        return NULL;

    freedv_data_set_header(fdc, fdc_header_bcast);
    memcpy(fdc->rx_header, fdc->tx_header, 8);
    
    return fdc;
}
```

dave-g8kbb, 2020-05-02:
Many thanks.
Indeed, a better solution. 
I think packet_tx_size is already being set by the line "fdc->packet_tx_size = 0;" in freedv_data_channel_create().
Did not know if there was any significant time overhead. I'll have a quick look round for any more this evening if I get chance...
Regards
Dave

srsampson, 2020-05-02:
> I think packet_tx_size is already being set by the line "fdc->packet_tx_size = 0;"

You're right

drowe67, 2020-05-02:
Good work @dave-g8kbb ad @srsampson :+1: 

Could one of you pls submit a Pull Request for the fix?

dave-g8kbb, 2020-05-02:
Happy to try - not done one before....

dave-g8kbb, 2020-05-02:
Have done pull request - first one I've done so apologies in advance :-)

srsampson, 2020-10-07:
Merged

closed 2020-10-07 by srsampson