drowe67/codec2-dev #109: Memory Leak Issues

drowe67, 2020-05-06:
@dave-g8kbb has kindly performed a code review and has the following suggestions:

https://github.com/drowe67/codec2/files/4586756/problems.in.codec2.4.may.2020.pdf

dave-g8kbb, 2020-05-09:
silly question - how do I make additional changes? I saved it as I needed to pop out but can't see how to edit it.
Dave

drowe67, 2020-05-09:
So you've checked in some more code?  I think you need to open a new pull request using the same branch.

hobbes1069, 2020-05-09:
You should just be able to commit the changes to your local fork/branch and then assuming you followed the typical workflow "git push origin <branch>"

dave-g8kbb, 2020-05-11:
Hi @drowe67 
I was still suffering problems and think I've found another one.
At line 226 of freedv_api.c in freedv_open_advanced(), f->ofdm is initialised when in 700D mode.
        f->ofdm = ofdm_create(ofdm_config);
        assert(f->ofdm != NULL);
        FREE(ofdm_config);

The next time it is touched is later in the function at line 302
	#ifndef __EMBEDDED__
        	/* tx BPF off on embedded platforms, as it consumes significant CPU */
        	ofdm_set_tx_bpf(f->ofdm, 1);
	#endif

f->ofdm will point to allocated memory for 700D mode or to NULL for all other modes.
However ofdm_set_tx_bpf() will access memory pointed to by ofdm even if it is NULL.
	void ofdm_set_tx_bpf(struct OFDM *ofdm, bool val) {
    		//fprintf(stderr, "ofdm_set_tx_bpf: val: %d ofdm_tx_bpf: %p \n", val, ofdm->ofdm_tx_bpf);
    		if (val == true) {
        		if (ofdm->ofdm_tx_bpf == NULL)
             			allocate_tx_bpf(ofdm);
        		ofdm->tx_bpf_en = true;
    		}
    		else {
        		if (ofdm->ofdm_tx_bpf != NULL)
            		deallocate_tx_bpf(ofdm);
        		ofdm->tx_bpf_en = false;
    		}
	}

A simple fix is to make the call to ofdm_set_tx_bpf() conditional on f->ofdm not being NULL however this precedes the setting of ofdm for 2020 mode which immediately follows the call in freedv_open_advanced(), so as well as making the call conditional you might want to consider moving it until after the 2020 initialisation when ofdm is allocated.

Shall I update that document and delete the incomplete code that I checked in and start again?

Regards
Dave

drowe67, 2020-05-11:
Hi @dave-g8kbb :smile:.  So what sort of problems are you encountering while using 700D?  Curious as this code is in daily use on several platforms with no reports of stability issues.  Some sort of corner case?

BTW we have a ctest for 700D that checks for memory leaks:
```
~/codec2/build_linux$ ctest -R test_memory_leak_FreeDV_700D_tx -V
```
IIRC the TX BPF is on by default so it should be exercising the code you mentioned above.  Perhaps the issue you have is something to do with multiple instances?

Re your report above, I can see the call to ```ofdm_set_tx_bpf(f->ofdm, 1);``` wrapped in the ```if (FDV_MODE_ACTIVE( FREEDV_MODE_700D, mode) )``` if-then block.  ```f->ofdm``` hasn't been freed so it should be non-null.  I popped an assert in there:
```
assert(f->ofdm != NULL);
ofdm_set_tx_bpf(f->ofdm, 1);
```
 and it didn't fire for any non 700D modes, e.g.
```
./freedv_tx 800XA ../../raw/ve9qrp.raw - | ./freedv_rx 800XA - - -vv | aplay -f S16_LE
```
Pls let me know what you think.

dave-g8kbb, 2020-05-12:
Hi @drowe67 
Quisk wakes up in 700D mode and runs fine. The problems occurred when switching modes, when quisk closes its freedv channels and reopens in the new mode. I don't think it is multiple channel related as I changed the code back to 1 channel and it still happened. Initially (with the first 800XA problem and 2 channels) it would cause an instant crash when switching to 800XA hence I went looking specifically for that. Now, before the last change, it will run fine, but switching into another mode would then result in a random crash some time later - anything from a second or two to a minute or so. The symptoms of the crash would be a python debugger stack trace indicating a memory fault - typically a memory segmentation error - in a low level codec2 or (occasionally) system core function. Not always the same codec2 function. Some times the stack trace would lead back to a quisk function call for the rx functions but usually just to a random stack trace of python functions whose depth varied from a few rows (until it was truncated due to stack corruption) to dozens of rows back through windows dll and python calls - but always with memory corruption. So each time I've tried to follow the scant breadcrumbs the debugger gives or when bored of that gone looking at the code again searching for initialisation where either memory is not cleanly allocated or deallocated. There is no guarantee that it's not due to a fault in quisk or a python runtime fault of course, I have also been modifying quisk for the other modes and correcting mistakes I have made as I found them. Still working on it but seems to be getting more reliable.
Dave

drowe67, 2020-05-12:
Oh wow that does sound like a tough bug hunt for sure.  I don't have any similar reports from FreeDV running on other platforms, e.g. the freedv-gui (running on three operating systems) and sm1000 users change modes all the time with no issue that I am aware of. 

From my side - I will write a unit test that automatically cycles through modes and see if that brings out any problems.  I've also emailed Jim (Quisk author) re the issue.

dave-g8kbb, 2020-05-12:
It's been fun. I've enjoyed it. Good to close the loop with Jim - I've tried to keep him up to date with findings via the groups.io page he runs. As I have not managed to crash the program all evening now, I think I'll concentrate next on the quisk changes - I have been tweaking it to cater for the modem & speech sampling rate differences of the other modes but not finished it yet. Promised Jim the code as soon as I am not embarrased by my tweaks. 
We have club members here running smalldv, sm1000, freedv and  (in my case) quisk. A number of us were monitoring a daily 700D net in Wales this afternoon - the thing that impressed everyone was hearing first hand the recovered audio from signals barely scraping out of the noise and not copyable on ssb. Impressive.

dave-g8kbb, 2020-05-23:
Hi @drowe67 
Oops - Ignore that last one I posted above (at line 226 et seq) I was quite wrong. Apologies.
Think I've got quisk running now as good as I'm going to get it. I still get occasional crashes in 2020 rx mode - can be anything from a few minutes to a few hours later. Always in the same place. Buried in LPCNet in vec_avx.h at line 130 in function vec_tanh(). SIGSEGV segmentation fault. I think this implies a memory alignment error for an avx 256 bit register (not sure - outside my experience now) but as far as I can see the addresses of variables input to the function are aligned on 32 byte boundaries.
Why this happens no idea. 
Quisk now supports 800XA and 2020 and I've coded in 2400B but not tested it on TX yet.

Best regards
Dave

drowe67, 2020-05-23:
@dave-g8kbb Nice work :+1: 

There has been a lot of refactoring around freedv-api recently, I've just merged the latest version.  

While I haven't had any other reports of stability issues with 2020, I have discovered a memory leak in that mode - might be related to your problem, will look into that next weekend.

How is your club project going?  I'd be happy to write a blog post, and link to your project page if you like.

dave-g8kbb, 2020-05-23:
Just about to get back to that now I've gone as far as I can with quisk. Just sent updated files to James for his consideration. The segmentation error is most weird as that function as far as I can see uses memory allocated by a higher level calling function in LPCNet and not from addresses passed in. Most weird.
I saw you were updating files. I'll wait for things to stabilise before updating this end :-)

dave-g8kbb, 2020-05-23:
doh! happened again just as I was typing the above response.
Thread 8 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 14000.0xfec]
0x00000000010a0852 in vec_tanh (N=<optimized out>, x=<optimized out>, y=<optimized out>)
    at C:/msys64/home/Dave/LPCNet/src/vec_avx.h:130
130             Y = exp8_approx(X);
(gdb) where
#0  0x00000000010a0852 in vec_tanh (N=<optimized out>, x=<optimized out>,
    y=<optimized out>) at C:/msys64/home/Dave/LPCNet/src/vec_avx.h:130
#1  compute_activation (output=0xcf2057ac80, input=0xcf2057ac80, N=128,
    activation=<optimized out>) at C:/msys64/home/Dave/LPCNet/src/nnet.c:80
#2  0x00000000010a4be3 in compute_conv1d (layer=0x19d1fa0 <feature_conv1>,
    output=output@entry=0xcf2057ac80, mem=mem@entry=0x21bc5440870,
    input=input@entry=0xcf2057aae0) at C:/msys64/home/Dave/LPCNet/src/nnet.c:333
#3  0x000000000109eb20 in run_frame_network (lpcnet=lpcnet@entry=0x21bc5440870,
    condition=condition@entry=0xcf2057b3d0,
    gru_a_condition=gru_a_condition@entry=0xcf2057c1d0,
    features=features@entry=0xcf2057e830, pitch=pitch@entry=100)
    at C:/msys64/home/Dave/LPCNet/src/lpcnet.c:83
#4  0x000000000109efde in lpcnet_synthesize (lpcnet=0x21bc5440870,
    output=0x70aab174 <rx_channel+20>, features=0xcf2057e830, N=160, logmag=0)
    at C:/msys64/home/Dave/LPCNet/src/lpcnet.c:163
#5  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb)

drowe67, 2020-05-24:
@dave-g8kbb OK thanks lets keep an eye on that one. 

Perhaps we need to ensure memory is on the correct boundary.  Another possibility is the neural net received an out of range value that causes a table look up to fly off out of allocated memory - have hit that one before.

drowe67, 2020-05-24:
This NN stuff is basically compute a big matrix multiply, pass through a non-linear function (like tanh), feed that to the next matrix.  Repeat over really big matrices 1E6 times :grinning:

srsampson, 2020-05-24:
There's some weird vector loops in there like 0 to N-7 step 1. Odd steps can't be good on a 64 bit machine (maybe).

dave-g8kbb, 2020-05-27:
I modified smalldv to support the modes; seems to run fine on transmit on 2020 mode but surprise surprise runs out of steam on receive, even if the Pi4 is overclocked. Is it a case of wait for a pi5 or is there anything I can usefully do? Tops out on 54% CPU load clocked at 2GHz - I'm guessing it's not parallelisable / amenable to multi threading or is it?

srsampson, 2020-05-27:
Maybe swap out some of the math algorithms from this old NEON library:

http://projectne10.github.io/Ne10/

drowe67, 2020-05-27:
@dave-g8kbb yep - as per the FreeDV User Manual - 2020 is pretty experimental and at this stage we only support x86 with AVX.  There are a few NEON-optimised routines in there but still not fast enough on a Pi4.  Yes well spotted  - you guess re parallel operation of the LPCNet codec is correct.

drowe67, 2020-05-27:
I did some work on the Pi 3 in this area, a really tough chip to get decent DSP performance out of http://www.rowetel.com/?p=6577

dave-g8kbb, 2020-05-27:
Thanks chaps - will have a read. Got 2020 running in quisk (on my AMD FX-6100 system).

dave-g8kbb, 2020-05-29:
here is comparative timing for your neon tests for Pi4B (plain vanilla, no overclocking)
CFLAGS+=-mfpu=neon-fp-armv8 -march=armv8-a -mtune=cortex-a72 -mfloat-abi=hard

pi@raspberrypi:~/neon-dot-master $ make test
sum: 4e+09 FLOPS: 8e+09
sum: 4e+09 FLOPS: 8e+09
sum: 4.03116e+09 target cycles: 1e+09 FLOPS: 8e+09
sum: 4.03116e+09 target cycles: 1e+09 FLOPS: 8e+09
FLOPS: 4e+09
sum: 1e+06 target cycles: 1e+09 FLOPS: 8e+09
grep cycles dot_log.txt
     3,000,568,780      cycles:u                                                    
     1,000,756,041      cycles:u                                                    
     3,822,566,727      cycles:u                                                    
     3,163,214,632      cycles:u                                                    
     4,911,232,369      cycles:u                                                    
     3,166,507,579      cycles:u

drowe67, 2020-05-29:
@dave-g8kbb I had a very unpleasant two weeks trying to that chip to play nice with DSP - scouring a bunch of data sheets.  Really poor documentation....

dave-g8kbb, 2020-05-29:
@drowe67 I have no illusions of success - you are *way* more experienced here than I ever will be but thought I'd take a peek as a learning exercise :-)

drowe67, 2020-05-30:
@dave-g8kbb actually that's why I like to leave a record of what I have done and where I struggled, provides a path for other to pick you if they want to play :smile:

dave-g8kbb, 2020-06-29:
I found the reason for the random crashes at vec_tanh as above. It's a weird one. My system's CPU is a 6 core AMD FX-6100. The code checks to see if AVX is supported and believes it is, and the code runs with avx commands, but AMD on their website say it is not. It seems therefore that there is support but LPCNet then crashes randomly after a short time, suggesting that there is something incomplete in its implementation in this CPU. More details here (as it's an LPCNet issue not a codec2 issue). 
https://github.com/drowe67/LPCNet/issues/32 
Dave

srsampson, 2020-10-07:
Problem moved to LPCNet

closed 2020-10-07 by srsampson