drowe67/codec2-dev #146: unrecognized command line option ‘-no-pie’

aviasd, 2020-09-13:
Hey,  
I am trying to install codec2 on AMI (AWS SageMaker), and these are the steps i've taken:  

```
sudo yum -y install cmake3
git clone https://github.com/drowe67/codec2
cd codec2
mkdir build_linux
cd build_linux
cmake3 ..
make
```

I took the instructions from [README.md](https://github.com/drowe67/codec2/blob/master/README.md)  
I get an error.  
This is the important part of the error:  
[image]

This is all the output:  
<details>
<summary>See Full Output</summary>
<p>

```
mkdir: cannot create directory ‘build_linux’: File exists
-- The C compiler identification is GNU 4.8.5
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- codec2 version: 0.9.2
-- freedv-gui current git hash: 55d7bb8d
-- Performing Test COMPILER_SUPPORTS_GNU11
-- Performing Test COMPILER_SUPPORTS_GNU11 - Success
-- Performing Test COMPILER_SUPPORTS_GNU99
-- Performing Test COMPILER_SUPPORTS_GNU99 - Success
-- Found Git: /usr/bin/git (found version "2.18.4") 
-- Codec2 current git hash: 55d7bb8d
-- Looking for include file stdlib.h
-- Looking for include file stdlib.h - found
-- Looking for include file string.h
-- Looking for include file string.h - found
-- Looking for __GNU_LIBRARY__
-- Looking for __GNU_LIBRARY__ - found
-- Looking for floor
-- Looking for floor - found
-- Looking for ceil
-- Looking for ceil - found
-- Looking for pow
-- Looking for pow - found
-- Looking for sqrt
-- Looking for sqrt - found
-- Looking for sin
-- Looking for sin - found
-- Looking for cos
-- Looking for cos - found
-- Looking for atan2
-- Looking for atan2 - found
-- Looking for log10
-- Looking for log10 - found
-- Looking for round
-- Looking for round - found
-- Looking for getopt
-- Looking for getopt - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
[... 337 lines cut ...]
make[1]: *** [unittest/CMakeFiles/ofdm_stack.dir/all] Error 2
make: *** [all] Error 2
```

</p>
</details>

I tried to update the gcc version but I get:  

```
Loaded plugins: dkms-build-requires, priorities, update-motd, upgrade-helper,
              : versionlock
Package gcc-4.8.5-1.22.amzn1.noarch already installed and latest version
Nothing to do
```

This is the machine I run:  

```
Amazon Linux AMI release 2018.03
```

The problem is with the `-no-pie` option, but I can not find a way to make it work in the AMI.   
Is there a way to get around it?   
Thank you,

hobbes1069, 2020-09-13:
Well it appears you're running a pretty old version of gcc (4.8.5) which does not recognize that option. The quick fix is likely to manually remove those flags. Long term we might be able to do some version detection of gcc and not add them if it's too old.

```
$ grep -r "no-pie"
unittest/CMakeLists.txt:    target_link_libraries(ofdm_stack function_trace m -no-pie "-Wl,-Map=ofdm_stack.map")
unittest/CMakeLists.txt:    target_compile_options(ofdm_stack PUBLIC -fstack-usage -finstrument-functions -no-pie)
unittest/CMakeLists.txt:    target_link_libraries(ofdm_stack function_trace m -no-pie)
unittest/CMakeLists.txt:    target_compile_options(ofdm_stack PUBLIC -finstrument-functions -no-pie)
```

srsampson, 2020-09-13:
Amazon Linux 2 is current. The last is End of Life in December.

srsampson, 2020-10-06:
Obsolete Operating System and environment

closed 2020-10-06 by srsampson

h1z1, 2021-01-24:
For what it's worth, CMake can take care of it for you.   Version 3.14 fixed a bug where -pie wasn't added when setting POSITION_INDEPENDENT_CODE otherwise the default is false.