drowe67/codec2-dev #148: Use of Codec2 on DMR

rogerclarkmelbourne, 2020-10-04:
Hi David

I'm interested in experimenting with Codec2 on DMR, but I don't know whether it could be configured to encode the correct number of bytes per frame

As you probably know DMR audio frames are 60mS long, and the compressed audio is 27 bytes long per frame.

As DMR supports 2 TimeSlots, those 27 bytes are sent in 30ms (plus some other addressing and flag and error correction data etc)

However, I think the key thing is whether codec 2 could generate 27 bytes per frame

The lowest audio sample rate I have available is 8khz, and I think higher rates would cause unnecessary CPU overhead on the encode and decode.

I'm willing to have a go at changing the code, but I have no idea where to start or whether this is even possible to configure the codec to generate 27 bytes output for 480 bytes input.

BTW.
I will need to compile it for ARM Cortex M4 for deployment onto a NXP MK22 MCU.
I'm also assuming that the code is in C and not C++ as I need to link into existing C code.

srsampson, 2020-10-04:
An alternative way of thinking, is to completely bypass the standard vocoder and send the voice packets via the data interface. Wrap the vocoder data in TCP/IP and badda-boom...

Yes, it is written in C

drowe67, 2020-10-04:
So that's 27*8/0.06 = 3600 bits/s. You don't really need to modify Codec 2, rather you just some code to buffer/pad the Codec 2 bit stream (say 3200 bits/s mode, or maybe 2400 with some FEC) to produce DMR frames.

rogerclarkmelbourne, 2020-10-04:
@drowe67 

Thanks

What sample rate does the codec expect? 8kHz ?    
I presume this is important in terms of the way it works.
Or doesn't this matter ?

I see on your website you tested at 3200bps, so perhaps that would be a better fit for DMR?

Re: Adding FEC

I'll need to take a look at the existing DMR FEC.  

From what I recall, I can't simply send any bytes I like as the audio data, because DMR network systems expect specific FEC to have been applied. 
e.g. One time I did try sending 0x00, 0x01 ... 0x26 as the audio stream, while I was debugging some code, and MMDVMHost mangled the data, because it seems to check and correct errors in the audio bytes.

I'll try to isolate the existing AMBE codec , encoding and decoding from the DMR FEC, when I have time, and then get back to you on this one

Thanks

rogerclarkmelbourne, 2020-10-04:
@ srsampson 

I don't think it would be a good idea to send this as DMR data e.g. like an SMS, because MMDVMHost and BM / MARC treat data packets separately in the way their UI displays the  packets

So in PiStar etc, the display would probably freak out, if it thinks the data is thousands of SMSs

darksidelemm, 2020-12-16:
Note that some work has been started on this, e.g. getting codec2 running on a MD380: https://github.com/TheSkorm/md380-codec2

Non-functional, but it's been thought about previously.

closed 2021-01-14 by drowe67