drowe67/codec2-dev #165: Codec2 not building on macOS Big Sur (11.0)

tmiw, 2021-01-03:
Just pulled the latest master and got the following:

```
[ 29%] Building C object src/CMakeFiles/fmfsk_mod.dir/fmfsk_mod.c.o
/Users/mooneer/devel/freedv-gui/codec2/src/fmfsk_mod.c:47:9: error: implicitly declaring library function 'exit' with type
      'void (int) __attribute__((noreturn))' [-Werror,-Wimplicit-function-declaration]
        exit(1);
        ^
/Users/mooneer/devel/freedv-gui/codec2/src/fmfsk_mod.c:47:9: note: include the header <stdlib.h> or explicitly provide a
      declaration for 'exit'
/Users/mooneer/devel/freedv-gui/codec2/src/fmfsk_mod.c:51:10: error: implicit declaration of function 'atoi' is invalid in
      C99 [-Werror,-Wimplicit-function-declaration]
    Fs = atoi(argv[1]);
         ^
/Users/mooneer/devel/freedv-gui/codec2/src/fmfsk_mod.c:76:24: error: implicitly declaring library function 'alloca' with
      type 'void *(unsigned long)' [-Werror,-Wimplicit-function-declaration]
    bitbuf = (uint8_t*)alloca(sizeof(uint8_t)*fmfsk->nbit);
                       ^
/Users/mooneer/devel/freedv-gui/codec2/src/fmfsk_mod.c:76:24: note: include the header <stdlib.h> or explicitly provide a
      declaration for 'alloca'
3 errors generated.
```

I'll try adding stdlib.h to that file and see if I get further.

drowe67, 2021-01-03:
@tmiw I think @ra1nb0w has addressed that one in https://github.com/drowe67/codec2/pull/164

tmiw, 2021-01-03:
@drowe67, oops. Never mind on this one then 👍

closed 2021-01-03 by tmiw