drowe67/codec2-dev #186: "In-place" FFT codepath is not used

nevercast, 2021-05-17:
Hello, I think this line 

https://github.com/drowe67/codec2/blob/master/src/codec2_fft.c#L141

Is meant to read

```
if (cfg->nfft <= 512)
```

Since nfft is multiplied by sizeof during the memcopy, it's safe that the 512 buffer and 512 condition be directly compared without converting into bytes.

nevercast, 2021-05-17:
I see this is in direct contrast with the original commit https://github.com/drowe67/codec2/commit/413faeb29ceeb7167acf2a8b50e30ddf32d135d2 

So perhaps I'm mistaken in this regard.

drowe67, 2021-05-17:
@nevercast - thanks.  :thinking: must admit I'm not sure myself.  @db4ple do you have any thoughts on this one?

nevercast, 2021-05-17:
I was almost 100% confident until I noticed it was less of a typo and more an intentional commit.

drowe67, 2021-05-17:
@nevercast https://github.com/drowe67/codec2/pull/187

db4ple, 2021-05-17:
Hi, reading the comment I made I can clearly see that @nevercast is right. If it fits into the 512 element long buffer in allocated on stack, we want to use it. However, in any case, we should move the allocation of `in` inside the `if``.  
So this code needs fixing to make sense at all.

drowe67, 2021-05-17:
Fixed in #187 thanks @nevercast and @db4ple

closed 2021-05-17 by drowe67