drowe67/codec2-dev #189: FreeDV API uniform peak Tx levels

drowe67, 2021-05-24:
When using FreeDV the Tx drive is adjusted so that the ALC is just being nudged.  At present different modes require different drive levels.

I think we want to make sure the peak level is the same for all modes. It gets a bit complicated with compression (clipping), but I think we want the same Tx waveform peak level with compression on or off. A ctest could confirm this.  Suggest +/-16383 as the peak level (perhaps mandated by a clipper for any stray peaks).

tmiw, 2022-04-10:
As a temporary workaround, would it be beneficial to emulate this at the freedv-gui level? It could potentially be done in a similar fashion to the squelch indication where the setting for TX Attenuation is for the mode with the least amount of drive required (and other modes just get a certain dB level added to that setting).

drowe67, 2022-04-10:
Probably better just to fix it at the codec2 level, the 2020 modes have already been brought into line with https://github.com/drowe67/codec2/pull/274.  700C and 1600 would be straight fwd to adjust.

DG9BFC, 2022-04-10:
if needed i can make a list what settings in what mode is how high SUM of output level
to compare modes (what mode works how well) i would like to have all modes show the same level (or nearly)
so ... if you need that list ... just ask
i have a monitor pin on my amp that shows roughly the output power (without looking how high each different peak is but with summing up ALL bw of that signal) ... say you have an amp that can push out 100w ... now lower the level of each mode that it shows ... say 10 w (or maybe 20?!?) ... does not matter much ... just set them all to the same POWER IN SUM ... not POWER IN PEAKS ... that way we could do a "fair" test of the modes ... and switching from one to another would give roughly the same level  towards antenna (plusminus a bit)

drowe67, 2022-04-11:
We'll be normalizing peak power at the FreeDV API output (16 bit Tx samples).

For a given peak power, the PAPR and hence RMS power is set by the waveform design.  As we use different waveforms for each mode, it is not possible to normalize the average power. For example a FSK mode will always have a a 0dB PAPR, QPSK modes a few dB less.

closed 2022-07-31 by drowe67

DG9BFC, 2022-07-31:
why not make "roughly" the average power the same ?? cause that is what users see on their watt meter?!?

tmiw, 2022-08-01:
> why not make "roughly" the average power the same ?? cause that is what users see on their watt meter?!?

My understanding is that average power is inherently a property of the mode, so changing that would cause problems. @drowe67 can clarify if I'm not remembering correctly.

drowe67, 2022-08-01:
"Why not normalise to average power" is a good question.  Couple of reasons:
1. We need to set the peak level to avoid triggering ALC, as we don't want the power varying with peaks and troughs of the waveform.  For example if we trigger the ALC an occasional peak (common in OFDM) will push the PA gain and hence RMS power down.
2. The waveforms vary in Peak to Average Power Ratio (PAPR or Crest Factor) so adjusting for the same average power across waveforms would mean lowering the average level so that the highest PAPR waveform doesn't trigger the ALC.  This might means a backoff of 10-15dB. So most waveforms would have very low average power compared to what your Tx can achieve.  You would have to adjust drive for every waveform to get the maximum out of that waveform, or put up with a consistent (but low) average power.
3. To maximimse SNR at the receiver we want to maximise average power at the Tx.  This happens with the combination of low PAPR and driving the PA to it's peak.
4. Hence I've adjusted the drive such that the peaks are the same across waveforms.  Lower PAPR waveforms will have higher average powers (maximising performance), we will be getting the highest average power possible for each waveform, and won't have to adjust the Tx drive level.