drowe67/codec2-dev #191: Fix stm32 unit tests

drowe67, 2021-06-01:
WIP here https://github.com/drowe67/codec2/pull/190

drowe67, 2021-06-01:
In #190 @timw commented:
> Should we look into having the STM32 code build as part of the existing GitHub actions? Might catch issues like this sooner in the future, anyway.

@tmiw yes that would be very useful.  @db4ple did some fine work on this a few years ago, when we first set up the CI (using Travis at the time). Couple of issues:
1. The Standard Peripheral Library requires a human filling in a form to download it.  IIRC this might be a license issue.
1. Running the ctests requires a stm32 developer kit, so can't be run on the GitHub actions servers alone.  They are also pretty slow (> 10 minutes).  However just the build alone would catch many issues.

tmiw, 2021-06-02:
@drowe67, since I just downloaded the SPL, I took a look at the license PDF included in it. The important part seems to be the following:

[image]

Based on that, I don't think including it in a GitHub action would be problematic unless Codec2 ends up being relicensed under a proprietary license. :) That said, since the UT do take quite a while to run, it's probably better to run them on some other machine somewhere (and quite possibly only on timed intervals using master, not after every pull request).

drowe67, 2021-06-02:
Thanks @tmiw.  It seems their main concern ("Product" definition) is that the code is used just for STM devices, but it looks like it is OK for us to redistribute the tarball.  So to bypass the login page and automate a build, we would need to cache the SPL tarball somewhere.

Re unit tests, the other problem @db4ple grappled with was how to arrange for GitHub/Travis to access remote machines (which had the stm32 Discovery development system hanging off it) using SSH keys.  Automation of that (such that anyone pushing to GitHub triggered the stm32 tests on a remote machine) wasn't possible at the time with Travis as there was no way to securely store a key.  So I've just been manually running the tests/maintaining the code every now and then.

donmr, 2021-06-02:
Someone could set up a machine in their lab to pull master and run the 
tests on some schedule.

drowe67, 2021-06-02:
> Someone could set up a machine in their lab to pull master and run the tests on some schedule.

Hello @donmr :slightly_smiling_face: As an alternative to a schedule, some automation could make the test machine pull and ctest only when there are changes, e.g. by polling codec2/master for a change in the git hash.

@tmiw - a few years ago Don wrote the fine unit test framework for the stm32, as well as implementing the 700D port

tmiw, 2021-06-03:
I have a VM server here. Let me see if I can at least get a VM on it to fire a build/test cycle in response to an SSH request. If so, then it should be straightforward to trigger the request using a [GitHub webhook](https://stackoverflow.com/questions/26914819/create-a-github-webhook-for-when-a-pull-request-is-accepted-merged-to-master).

tmiw, 2021-06-03:
So yeah, looks like I can kick off a build over ssh no problem:

```
HT-TM05:pihpsdr mooneer$ ssh freedvbuild "nohup /home/mooneer/codec2_stm32_build.sh >/dev/null 2>/dev/null &"
HT-TM05:pihpsdr mooneer$
```

Failing, of course, since #190 hasn't been merged yet. Gmail is also throwing the failure email that it's sending into my Spam folder, which isn't helpful. Here's the script I'm using (email address redacted):

```
#!/bin/bash

export PATH=$HOME/gcc-arm-none-eabi-8-2018-q4-major/bin:$PATH

RESULT=$((rm -rf codec2 && \
git clone https://github.com/drowe67/codec2.git && \
cd codec2/stm32 && \
git checkout master && \
mkdir build_stm32 && \
cd build_stm32 && \
cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/STM32_Toolchain.cmake -DPERIPHLIBDIR=~/STM32F4xx_DSP_StdPeriph_Lib_V1.8.0 .. && \
make VERBOSE=1) 2>&1)

if [ $? -ne 0 ]; then
    /usr/sbin/sendmail -t <<EOF
To: [email]
Subject: [Codec2] Build failed

$RESULT
EOF
fi
```

drowe67, 2021-06-03:
Nice work @tmiw - yes I received the email as well :slightly_smiling_face:

tmiw, 2021-08-09:
@drowe67, seems like this can be closed now?

closed 2021-08-09 by drowe67