drowe67/codec2-dev #199: [700C] rx_filter_coh() potentially requires optimization

tmiw, 2021-07-24:
700C uses a significant amount of CPU on my system for receive (2019 MacBook Pro using an 8 core 2.3GHz i9) compared to other modes--even 2020. Profiling with the built-in Instruments tool on macOS resulted in the following:

[image]

Updating rx_filter_cox() per [this patch](https://github.com/drowe67/codec2/files/6872942/Codec2.Optimization.txt) helps only slightly, as shown here:

[image]

Anything else that could be tried here optimization-wise? For instance, I wonder if this could potentially use AVX (on supported systems).

drowe67, 2021-07-24:
Yes the cohpsk modem *acquisition* is fairly heavy in CPU, compared to the other modems.  This would be noticeable in freedv-gui multi-rx mode, when many modem instances are running in acquisition in parallel. 

When in sync the CPU is more moderate, once synced the LPCNet decoder used in 2020 uses vastly more CPU. 

Not sure if this is an actual *issue* - the 700C and cohpsk has been in wide use for 5 years, and a variety of platforms.  More optimisation is always welcome, as long as it doesn't break anything.

There was a bit of work done on this recently, in order to get it running on a Pi 3 https://github.com/drowe67/codec2/pull/185.  This PR also shows a good way to test 700C acquisition.

tmiw, 2021-07-24:
Speaking of the Pi, I'm running some ARM64 tests with the mentioned PR (compared to master) and will update the PR with those results shortly. 👍

drowe67, 2021-07-24:
I recall two platforms were this might help:
1. The stm32 if we wanted to run 700C on that. 
1. 700C is broken on on Flex as they use (last I heard) a fairly low end ARM machine for third party waveforms, and the CPU is too high.  A lesson there too - the original port of 700C to the Flex introduced a subtle bug into 700C (on all platforms), and was only noticed several years later.  ctests we have now should help prevent that.

tmiw, 2021-07-24:
> I recall two platforms were this might help:
> 
> 1. The stm32 if we wanted to run 700C on that.
> 2. 700C is broken on on Flex as they use (last I heard) a fairly low end ARM machine for third party waveforms, and the CPU is too high.  A lesson there too - the original port of 700C to the Flex introduced a subtle bug into 700C (on all platforms), and was only noticed several years later.  ctests we have now should help prevent that.

Depending on the optimizations needed, 32 bit ARM may be more problematic. I posted more in the linked PR.

srsampson, 2021-07-25:
The filter is used to work with 8 kHz or higher rate soundcards. However on firmware, you can probably just use 7500, and not need the conversion. It's just a simple A/D and D/A.

drowe67, 2021-07-25:
> The filter is used to work with 8 kHz or higher rate soundcards. However on firmware, you can probably just use 7500, and not need the conversion. It's just a simple A/D and D/A.

Hi Steve :slightly_smiling_face: IIRC this filter is the raised root cosine filter that runs at 7500 Hz; it's necessary for a parallel tone modem like cohpsk and the fdmdv (FreeDV 1600).  The filter you mentioned changes the cohpsk sample rate to 8000 Hz (I think) and is in the FreeDV API source.  That (and the various Tx BPFs used for OFDM modes) could probably benefit from some vectorisation too :+1:

srsampson, 2021-07-25:
> > The filter is used to work with 8 kHz or higher rate soundcards. However on firmware, you can probably just use 7500, and not need the conversion. It's just a simple A/D and D/A.
> 
> Hi Steve  IIRC this filter is the raised root cosine filter that runs at 7500 Hz; it's necessary for a parallel tone modem like cohpsk and the fdmdv (FreeDV 1600). The filter you mentioned changes the cohpsk sample rate to 8000 Hz (I think) and is in the FreeDV API source. That (and the various Tx BPFs used for OFDM modes) could probably benefit from some vectorisation too

Ah! My bad.

drowe67, 2021-08-20:
@tmiw I think you addressed this one with https://github.com/drowe67/codec2/pull/200

closed 2021-08-20 by drowe67

tmiw, 2021-08-21:
@drowe67, there were maybe one or two more "nice to haves" but those are definitely low priority (we aren't currently targeting much slower than, say, a rPi3). 👍

drowe67, 2021-08-21:
@tmiw it would be useful to have a brainstorm lareawish list on GitHub, so we can separate actual issues (ie bugs that need fixing, or mainstream development we have committed to) from the nice to haves, and discussion points.

Maybe we could use https://github.com/drowe67/codec2/discussions for the non-Issue topics?

tmiw, 2021-08-21:
Yep, just added #221 👍