drowe67/codec2-dev #201: Error compiling for ESP32

happytm, 2021-07-29:
I am trying to use Codec2 library for ESP32 wemos lolin32 board usin code and this library from following repo:

https://github.com/deulis/ESP32_Codec2

The owner of above library is not responsive so I am posing a question here.

When I try to compile above sketch under arduino it give me following message:

```
C:\Arduino_1.8.12\portable\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware C:\Arduino_1.8.12\portable\arduino-nightly\hardware -hardware C:\Users\WinCycle\AppData\Local\Arduino15\packages -hardware C:\Users\WinCycle\Documents\Arduino\hardware -tools C:\Arduino_1.8.12\portable\arduino-nightly\tools-builder -tools C:\Arduino_1.8.12\portable\arduino-nightly\hardware\tools\avr -tools C:\Users\WinCycle\AppData\Local\Arduino15\packages -built-in-libraries C:\Arduino_1.8.12\portable\arduino-nightly\libraries -libraries C:\Users\WinCycle\Documents\Arduino\libraries -fqbn=esp32:esp32:lolin32:FlashFreq=80,PartitionScheme=default,CPUFreq=80,UploadSpeed=921600 -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\WinCycle\AppData\Local\Temp\arduino_build_140218 -warnings=all -build-cache C:\Users\WinCycle\AppData\Local\Temp\arduino_cache_128957 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose C:\Users\WinCycle\Documents\Arduino\ESP32_Codec2-master\ESP32_Codec2\ESP32_codec2.ino
C:\Arduino_1.8.12\portable\arduino-nightly\arduino-builder -compile -logger=machine -hardware C:\Arduino_1.8.12\portable\arduino-nightly\hardware -hardware C:\Users\WinCycle\AppData\Local\Arduino15\packages -hardware C:\Users\WinCycle\Documents\Arduino\hardware -tools C:\Arduino_1.8.12\portable\arduino-nightly\tools-builder -tools C:\Arduino_1.8.12\portable\arduino-nightly\hardware\tools\avr -tools C:\Users\WinCycle\AppData\Local\Arduino15\packages -built-in-libraries C:\Arduino_1.8.12\portable\arduino-nightly\libraries -libraries C:\Users\WinCycle\Documents\Arduino\libraries -fqbn=esp32:esp32:lolin32:FlashFreq=80,PartitionScheme=default,CPUFreq=80,UploadSpeed=921600 -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\WinCycle\AppData\Local\Temp\arduino_build_140218 -warnings=all -build-cache C:\Users\WinCycle\AppData\Local\Temp\arduino_cache_128957 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose C:\Users\WinCycle\Documents\Arduino\ESP32_Codec2-master\ESP32_Codec2\ESP32_codec2.ino
Using board 'lolin32' from platform in folder: C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
Using core 'esp32' from platform in folder: C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
Warning: Board breadboard:avr:atmega328bb doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA328BB
cmd /c if exist "C:\\Users\\WinCycle\\Documents\\Arduino\\ESP32_Codec2-master\\ESP32_Codec2\\partitions.csv" copy /y "C:\\Users\\WinCycle\\Documents\\Arduino\\ESP32_Codec2-master\\ESP32_Codec2\\partitions.csv" "C:\\Users\\WinCycle\\AppData\\Local\\Temp\\arduino_build_140218\\partitions.csv"
cmd /c if not exist "C:\\Users\\WinCycle\\AppData\\Local\\Temp\\arduino_build_140218\\partitions.csv" copy "C:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\partitions\\default.csv" "C:\\Users\\WinCycle\\AppData\\Local\\Temp\\arduino_build_140218\\partitions.csv"
Detecting libraries used...
"C:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bootloader_support" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bt" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/coap" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/console" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/driver" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/efuse" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp-tls" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp32" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_event" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_client" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_server" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_ota" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_server" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_websocket_client" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/espcoredump" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/ethernet" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/expat" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/fatfs" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freemodbus" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freertos" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/heap" "-IC:\\Users\\WinCycle\\AppData\\Local\\Arduino15\\packages\\esp32\\hard [... cut ...]
                                  ^
cc1.exe: some warnings being treated as errors
Using library SPI at version 1.0 in folder: C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SPI 
Using library LoRa at version 0.8.0 in folder: C:\Users\WinCycle\Documents\Arduino\libraries\LoRa 
Using library codec2 in folder: C:\Users\WinCycle\Documents\Arduino\libraries\codec2 (legacy)
exit status 1
Error compiling for board WEMOS LOLIN32.
```

Is there any solution to this problem?

Thanks.

srsampson, 2021-07-29:
Maybe change line 58 of phaseexp.c from:

char c;

to:

unsigned char c;

Doesn't seem to like it being a 'signed char' inside the ctype.h macro decode.

happytm, 2021-07-29:
Wow. That was fast reply and solution you provided worked.

Thank you very much.

deulis, 2021-07-29:
Wow, you can't wait two hour? hahaha. Anyway Thanks I will double check the solution. I hope everything is working fine.

tmiw, 2021-07-30:
On that note, there were significant changes in the main Codec2 library since the GitHub author mentioned above worked on it, so it might be worthwhile to integrate those changes at some point. 

Additionally, it may be a good idea to try to compile with -Wall -Werror and see how many failures actually happen. I know I was getting a bunch of "uninitialized value" errors from Codec2 when I was trying to use Valgrind to debug a freedv-gui memory leak recently, anyway.

deulis, 2021-07-30:
Yes, there are many changes from the version I am using and the latest one. My first attempt was using the last version but a got too many errors and compilation problems. So I finally used an old version that I had from another Android and Linux project. Here the amazing things is that the codec2 can work with an ESP32, but I know there is a lot of work to do.

drowe67, 2021-07-30:
Well done @deulis on getting Codec 2 to run on an ESP32.

Note to self - it would be nice to have some unit tests to support testing Codec 2 ports to other machines.  This is tricky, as different FPUs (e.g. x86 and stm32) give different results, so bit exact tests won't work.  Subjective listening tests can be unreliable with speech.  On the stm32 ctests we compare the average energy of output speech samples (with the same input speech sample) as a rough guide, this will show up any gross errors (like stack overflow causing a crash).  We could also have lower level unit tests, e.g compare pitch estimation contours, spectral magnitudes {Am} etc

deulis, 2021-07-31:
When I realized that the ESP32 has FPU my first attempt was made your fabulous Codec2 work on this, and it was very easy, so it show that you have a great code!.  Here also was used the ADC and DAC of the ESP32 and a LoRa radio for communications, all of them in the same module. But all the credit goes to you for making this wonderful open source project and dedicating so many hours to its maintenance.

drowe67, 2021-08-01:
> Additionally, it may be a good idea to try to compile with -Wall -Werror and see how many failures actually happen. I know I was getting a bunch of "uninitialized value" errors from Codec2 when I was trying to use Valgrind to debug a freedv-gui memory leak recently, anyway.

We run a set of valgrind tests in the Codec 2 ctests.  IIRC these have occasionally detected unitialised memory errors (which were fixed), as well as memory leaks.  Yes it would be good to enable warnings and reconcile any differences between freedv-gui and codec2 running stand alone.

tmiw, 2021-08-02:
BTW, I just ran with -Wall and created a new issue with the warnings I found: https://github.com/drowe67/codec2/issues/202

drowe67, 2021-08-20:
Looks like this was solved above https://github.com/drowe67/codec2/issues/201#issuecomment-889492144

closed 2021-08-20 by drowe67