drowe67/codec2-dev #207: codec2 1.0.0 build does not install c2enc artifact

chenrui333, 2021-08-15:
👋  it looks like with the codec 2 1.0.0 release build, I could not find the c2enc got installed.

relates to https://github.com/Homebrew/homebrew-core/pull/83293

tmiw, 2021-08-15:
It doesn't look like any of the test apps install on "make install". @drowe67, which of them (if not all) should be installed?

Also, note to self: whatever we add will need to be added when creating test Debian packages as well.

srsampson, 2021-08-15:
I think most users don't want that low level stuff in their /usr while experimenters do.

Maybe have a "make install commandline" or some such.

drowe67, 2021-08-15:
IIRC a few years ago we made the call that just libcodec2 would be installed, as @srsampson says everything else is really for experimenters or demos.

@hobbes1069 can you recall where we landed with that decision?

If we get some feedback from end users on what they actually use codec 2 for, I'd be happy to add extra targets.  I've asked packaging maintainers this question in the past but they don't seem to have that information. So I just assume that most people who want codec2 packaged use it for freedv-gui.

I think this section of `codec2/src/CmakeLists.txt` sets up what is installed:
```
install(TARGETS codec2 EXPORT codec2-config
    LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR} COMPONENT lib
    ARCHIVE DESTINATION ${CMAKE_INSTALL_LIBDIR} COMPONENT lib
    RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR}
    PUBLIC_HEADER DESTINATION ${CMAKE_INSTALL_INCLUDEDIR}/codec2 COMPONENT dev
)
```

drowe67, 2021-08-15:
@chenrui333 - why do you want `c2enc` installed?  What is the use case?

hobbes1069, 2021-08-16:
I'm fine with them not being installed by default but we could create an option to install them. Currently they are unconditionally built, but not installed so we wrong twice :)

On a side note, we need to update the unit tests to be build dir agnostic. Probably easiest to make the scripts .in files and substitute the build directory using cmake config directives.

tmiw, 2021-08-16:
Perhaps there could be a "codec2-tools" package that people can install if they're interested? IIRC we already generate an lpcnet-tools Debian package when someone runs "cpack".

@hobbes1069, it probably wasn't wrong to build the tools by default (as some are used in the current unit tests), so no worries there. 👍

drowe67, 2021-08-16:
Hi @hobbes1069 

> On a side note, we need to update the unit tests to be build dir agnostic. Probably easiest to make the scripts .in files and substitute the build directory using cmake config directives.

I did try the script `.in` approach but found it awkward to debug/develop, as you need a cmake step when you modify the test script source, e.g. a bash script with cmake directives in it won't run stand-alone. However we have managed to make quite a few of the tests build dir agnostic using other techniques.

Can you pls tell us which unit tests aren't working for you due to build dir problems?  We've been fixing up a few with this issue as they appear on OSX (it's usually me making them build dir dependent out of habit!)  There are some other issues that can be tougher cross-platform, such as numerical/floating point stuff.

hobbes1069, 2021-08-16:
I may have forgotten a `git pull` when I was poking around. Trying test nows mostly seem to work. I'm getting a couple of failures and 117 never finished. PS shows tfifo using 100% of 2 cores. I left it going for 10 minutes. I'll do some more poking around and open a new issue.

drowe67, 2021-08-20:
No response for OP on why we need to consider installing c2enc.  Happy to repoen if there is end user interest in installing anything apart from libcodec2.

closed 2021-08-20 by drowe67