drowe67/codec2-dev #208: Including Codec 2 in an Android project

Jarvif, 2021-08-16:
Hi,drowe67:
When I integrate codec2 into Android NDK, I will be prompted with an error C: / users / jarvif / desktop / codec2ndk / APP / SRC / main / codec2 / SRC / OFDM. C: 2428: error: undefined reference to 'cabsf'. Do you know why? thank you! And someone test it in an Android NDK?

tmiw, 2021-08-16:
cabsf is [supposed to be built into the C library](https://en.cppreference.com/w/c/numeric/complex/cabs). I'm not familiar with the NDK but perhaps there needs to be something else linked in for Android to use complex numbers?

drowe67, 2021-08-17:
@Jarvif have you found a way to include `cabsf` on Android?

Jarvif, 2021-08-19:
> cabsf is [supposed to be built into the C library](https://en.cppreference.com/w/c/numeric/complex/cabs). I'm not familiar with the NDK but perhaps there needs to be something else linked in for Android to use complex numbers?

maybe,thanks!

drowe67, 2021-08-20:
This doesn't appear to be a codec2 issue, more a configuration issue for OP

closed 2021-08-20 by drowe67

Jarvif, 2021-08-21:
> This doesn't appear to be a codec2 issue, more a configuration issue for OP
hi，drowe67 and tmiw. Codec2 in Android,must requires ANDROID MINSDK API 23+. This is because Android's NDK does not support compiling various complex.h math functions on prior API versions.See the Bionic Release Status for details. Thank you very much!

Jarvif, 2021-08-21:
the link about Android bionic status.[https://android.googlesource.com/platform/bionic/+/master/docs/status.md](url)

srsampson, 2021-08-21:
> Do you know why?

We don't know why, but if you figure it out let us know!

Jarvif, 2021-08-21:
> > Do you know why?
> 
> We don't know why, but if you figure it out let us know!

 This is because Android's NDK does not support compiling various complex.h math functions on prior API versions.See the Bionic Release Status for details.the link about Android bionic status.https://android.googlesource.com/platform/bionic/+/master/docs/status.md

Jarvif, 2021-08-21:
> > Do you know why?
> 
> We don't know why, but if you figure it out let us know!

Codec2 in Android,must requires ANDROID MINSDK API 23+.