drowe67/codec2-dev #210: Multiple tests failing on Fedora x86_64

hobbes1069, 2021-08-16:
Most seem to fail for regex issues I need to investigate, but as mentioned 117 (tfifo) never finished.

```
$ grep "***Failed" results_codec2/1.0.0/1.fc36.2/build.log 
  2/117 Test   #2: test_fdmdv_16to8 ...............................***Failed  Required regular expression not found. Regex=[PASS
  3/117 Test   #3: test_fdmdv_16to8_short .........................***Failed  Required regular expression not found. Regex=[PASS
  4/117 Test   #4: test_CML_ldpcut ................................***Failed  Required regular expression not found. Regex=[Nerr: 0
  6/117 Test  #12: test_OFDM_modem_octave_port ....................***Failed  Required regular expression not found. Regex=[fails: 0
  7/117 Test  #13: test_OFDM_modem_octave_port_Nc_31 ..............***Failed  Required regular expression not found. Regex=[fails: 0
  8/117 Test  #14: test_OFDM_modem_octave_qam16_uncoded ...........***Failed  Required regular expression not found. Regex=[Pass
  9/117 Test  #15: test_OFDM_modem_esno_est_octave ................***Failed  Required regular expression not found. Regex=[AWGN Pass.*MPP Pass
 10/117 Test  #16: test_OFDM_modem_esno_est_c .....................***Failed  Required regular expression not found. Regex=[AWGN Pass.*MPP Pass
 11/117 Test  #18: test_OFDM_modem_octave_datac0_postamble ........***Failed  Required regular expression not found. Regex=[Pass
 12/117 Test  #19: test_OFDM_modem_burst_acq_port .................***Failed  Required regular expression not found. Regex=[PASS
 13/117 Test  #17: test_OFDM_modem_octave_burst_acq ...............***Failed  Required regular expression not found. Regex=[P.acq. = 1.00
 14/117 Test  #20: test_OFDM_modem_octave_datac0_mpp ..............***Failed  Required regular expression not found. Regex=[Pass
 15/117 Test  #21: test_OFDM_modem_octave_datac0_mpp_coded ........***Failed  Required regular expression not found. Regex=[Pass
 17/117 Test  #26: test_OFDM_modem_fading_BER .....................***Failed    0.00 sec
 19/117 Test  #27: test_OFDM_modem_phase_est_bw ...................***Failed    0.01 sec
 20/117 Test  #28: test_OFDM_modem_fading_DPSK_BER ................***Failed    0.01 sec
 23/117 Test  #32: test_OFDM_modem_datac0_octave_burst ............***Failed  Required regular expression not found. Regex=[Coded PER: 0.0000 Tpkts:     3
 24/117 Test   #7: test_FDMDV_modem_octave_port ...................***Failed  Required regular expression not found. Regex=[fails: 0
 25/117 Test  #33: test_OFDM_modem_datac1_octave ..................***Failed  Required regular expression not found. Regex=[Coded PER: 0.0000 Pckts:     4
 29/117 Test  #34: test_OFDM_modem_datac3_octave ..................***Failed  Required regular expression not found. Regex=[Coded PER: 0.0000 Pckts:     5
 32/117 Test   #6: test_codec2_700c_octave_port ...................***Failed  Required regular expression not found. Regex=[fails: 0
 36/117 Test   #8: test_COHPSK_modem_octave_port ..................***Failed  Required regular expression not found. Regex=[fails: 0
 51/117 Test  #62: test_freedv_api_rawdata_800XA ..................***Failed    0.01 sec
 56/117 Test  #66: test_memory_leak_FreeDV_1600_tx ................***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 57/117 Test  #68: test_memory_leak_FreeDV_700D_tx ................***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 59/117 Test  #70: test_memory_leak_FreeDV_700C_tx ................***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 60/117 Test  #67: test_memory_leak_FreeDV_1600_rx ................***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 61/117 Test  #72: test_memory_leak_FreeDV_FSK_LDPC_tx ............***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 62/117 Test  #73: test_memory_leak_FreeDV_DATAC0_tx ..............***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 63/117 Test  #69: test_memory_leak_FreeDV_700D_rx ................***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 64/117 Test  #71: test_memory_leak_FreeDV_700C_rx ................***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 65/117 Test  #74: test_memory_leak_FreeDV_DATAC1_tx ..............***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 66/117 Test  #75: test_memory_leak_FreeDV_700E_tx ................***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 67/117 Test  #76: test_memory_leak_FreeDV_2020_tx ................***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 68/117 Test  #29: test_OFDM_modem_time_sync_700D .................***Failed    0.95 sec
 69/117 Test  #79: test_codec2_mode_3200 ..........................***Failed    0.02 sec
 72/117 Test  #80: test_codec2_mode_2400 ..........................***Failed    0.03 sec
 73/117 Test  #81: test_codec2_mode_1400 ..........................***Failed    0.01 sec
 74/117 Test  #82: test_codec2_mode_1300 ..........................***Failed    0.02 sec
 75/117 Test  #83: test_codec2_mode_1200 ..........................***Failed    0.01 sec
 76/117 Test  #77: test_memory_leak_FreeDV_2020_rx ................***Failed  Required regular expression not found. Regex=[ERROR SUMMARY: 0 errors
 77/117 Test  #84: test_codec2_mode_700C ..........................***Failed    0.01 sec
 78/117 Test  #85: test_codec2_mode_450 ...........................***Failed    0.02 sec
 79/117 Test  #86: test_codec2_mode_450PWB ........................***Failed    0.02 sec
 81/117 Test  #90: test_fsk_lib ...................................***Failed  Required regular expression not found. Regex=[PASS
 82/117 Test  #91: test_fsk_modem_octave_port .....................***Failed  Required regular expression not found. Regex=[PASS
 83/117 Test  #30: test_OFDM_modem_time_sync_2020 .................***Failed    1.03 sec
 85/117 Test  #89: test_700c_eq ...................................***Failed    0.08 sec
 86/117 Test  #87: test_est_n0 ....................................***Failed    0.10 sec
 87/117 Test  #97: test_fsk_lib_4fsk_ldpc .........................***Failed  Required regular expression not found. Regex=[PASS
 98/117 Test #108: test_freedv_data_raw_ofdm_datac1 ...............***Failed    0.13 sec
100/117 Test #107: test_freedv_data_raw_ofdm_datac0 ...............***Failed    0.27 sec
102/117 Test #106: test_freedv_data_raw_ofdm_datac1_burst_file ....***Failed    0.46 sec
109/117 Test #105: test_freedv_data_raw_ofdm_datac0_burst_file ....***Failed    0.90 sec
```

hobbes1069, 2021-08-16:
I realized I didn't have octave-cli installed so I added to the build requirements of the package but it didn't seem to make much of a difference other than pulling in about ~600 packages as dependencies.

srsampson, 2021-08-16:
Make sure the library got updated in /usr/local/lib . After a build I have to manually update these files (libcodec2.so, and libcodec2.so.0.9 ) also can't hurt to do a sudo ldconfig.

srsampson, 2021-08-16:
That's the problem I have, but other people don't seem to have this problem on Ubuntu.

drowe67, 2021-08-16:
I just run the ctests on a local build without ever using `make install`.  @hobbes1069 perhaps you still have some missing packages, there is a list of the Octave and other packages required to support the ctests in the top level `README.md`
```
grep apt README.md
sudo apt install git build-essential cmake
sudo apt install octave octave-common octave-signal liboctave-dev gnuplot python3-numpy sox valgrind
```
`ctest -V test_name` on a particular test will give you more verbose information.

`codec2/.github/workflows/cmake.yml` shows how a complete instance for testing is built up.

A normal `tfifo` run;
```
avid@boomer:~/codec2/build_linux dr-trellis $ ctest -R test_fifo
Test project /home/david/codec2/build_linux
    Start 65: test_fifo
1/1 Test #65: test_fifo ........................   Passed    0.07 sec
```

hobbes1069, 2021-08-17:
Well technically in an RPM packages %install is run before %check but it's all being installed to a special buildroot for packaging so doesn't really matter. 

In any case, I know what a few problems are. I didn't have octave-signals which I've now added. The bigger problem is CML. It's not in Fedora and it's not something I can build on the fly for an official build.

hobbes1069, 2021-08-17:
I know it would be a bit of a pain, but could we test for the octave-cml plugin and if not present, just skip the tests that need it?

drowe67, 2021-08-17:
@hobbes1069 - how many tests are failing for you if CML is not present?

I recall some of the tests (e.g. `octave/tofdm.m`) can detect if CML is present and work around it.  However there are quite a few tests using the LDPC decoder in Octave source that CML provides.

Some other ideas:
1. Could we work out a neater way to build Octave CML support, to make it more portable?  The build instructions are at the top of `codec2/octave/ldpc.m`.  Would it help if some of the CML tar ball files were added as source files codec2?  It could then be built along with other unit tests binaries.
2. You can simply specify which tests to exclude using `ctest -E`

tmiw, 2021-08-17:
What about making CML a Git subproject? That would allow it to be only retrieved for the people who actually need it.

drowe67, 2021-11-11:
Hi @hobbes1069 where have we landed with this one?  Is it still considered an Issue (i.e. something that requires coding to fix) or do you have a satisfactory work around?

drowe67, 2021-12-04:
@hobbes1069 I'm closing this one for now, but feel free to re-open if you thinks it's an issue that can be addressed by any changes to codec2.

closed 2021-12-04 by drowe67