drowe67/codec2-dev #213: Built and installed codec2 library for Fedora 34 from current master #7e5403c - crash immediately on modem start

Tyrbiter, 2021-08-20:
gdb output shown below:

Thread 1 "freedv" received signal SIGSEGV, Segmentation fault.
ofdm_set_verbose (ofdm=0x0, level=-1) at /usr/src/debug/codec2-1.0.0-1.fc34.2.x86_64/src/ofdm.c:971
971	void ofdm_set_verbose(struct OFDM *ofdm, int level) { ofdm->verbose = level; }
(gdb) thread apply all bt full

Thread 81 (Thread 0x7fffc37fe640 (LWP 227978) "freedv"):
#0  0x00007ffff571da8a in __futex_abstimed_wait_common64 (futex_word=futex_word@entry=0x5555557975c8, expected=expected@entry=0, clockid=clockid@entry=1, abstime=abstime@entry=0x7fffc37fda70, private=private@entry=0, cancel=cancel@entry=true) at ../sysdeps/nptl/futex-internal.c:74
        sc_cancel_oldtype = 0
        sc_ret = <optimized out>
        clockbit = 0
        err = <optimized out>
        op = 137
#1  0x00007ffff571daef in __GI___futex_abstimed_wait_cancelable64 (futex_word=futex_word@entry=0x5555557975c8, expected=expected@entry=0, clockid=clockid@entry=1, abstime=abstime@entry=0x7fffc37fda70, private=private@entry=0) at ../sysdeps/nptl/futex-internal.c:123
#2  0x00007ffff5717892 in __pthread_cond_wait_common (abstime=0x7fffc37fda70, clockid=1, mutex=0x555555da2d80, cond=0x5555557975a0) at pthread_cond_wait.c:504
        spin = 0
        buffer = {__routine = 0x7ffff5717040 <__condvar_cleanup_waiting>, __arg = 0x7fffc37fd990, __canceltype = -134440518, __prev = 0x0}
        cbuffer = {wseq = 0, cond = 0x5555557975a0, mutex = 0x555555da2d80, private = 0}
        err = <optimized out>
        g = 0
        flags = <optimized out>
        g1_start = <optimized out>
        maxspin = 0
        signals = <optimized out>
        result = 0
        wseq = 0
        seq = 0
        private = 0
#3  __pthread_cond_clockwait (abstime=0x7fffc37fda70, clockid=1, mutex=0x555555da2d80, cond=0x5555557975a0) at pthread_cond_wait.c:672
#4  __pthread_cond_clockwait (cond=0x5555557975a0, mutex=0x555555da2d80, clockid=1, abstime=0x7fffc37fda70) at pthread_cond_wait.c:660
#5  0x000055555560da8d in std::__condvar::wait_until(std::mutex&, int, timespec&) (__abs_time=..., __clock=1, __m=..., this=0x5555557975a0) at /usr/include/c++/11/bits/std_mutex.h:169
        __ts = {tv_sec = 45171, tv_nsec = 899140369}
        __my_lock = {_M_device = 0x0, _M_owns = false}
        lock = {_M_device = 0x555555797578, _M_owns = false}
#6  std::condition_variable::__wait_until_impl<std::chrono::duration<long, std::ratio<1l, 1000000000l> > >(std::unique_lock<std::mutex>&, std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&) (__lock=<optimized out>, __lock=<synthetic pointer>..., __atime=<optimized out>, this=0x5555557975a0) at /usr/include/c++/11/condition_variable:201
        __ts = {tv_sec = 45171, tv_nsec = 899140369}
        __my_lock = {_M_device = 0x0, _M_owns = false}
        lock = {_M_device = 0x555555797578, _M_owns = false}
#7  std::condition_variable::wait_until<std::chrono::duration<long, std::ratio<1l, 1000000000l> > >(std::unique_lock<std::mutex>&, std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&) (__atime=<optimized out>, __lock=<synthetic pointer>..., this=0x5555557975a0) at /usr/include/c++/11/condition_variable:111
        __my_lock = {_M_device = 0x0, _M_owns = false}
        lock = {_M_device = 0x555555797578, _M_owns = false}
#8  std::_V2::condition_variable_any::wait_until<std::unique_lock<std::mutex>, std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >(std::unique_lock<std::mutex>&, std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&) (__atime=<optimized out>, __lock=..., this=0x5555557975a0) at /usr/include/c++/11/condition_variable:336
        __my_lock = {_M_device = 0x0, _M_owns = false}
--Type <RET> for more, q to quit, c to continue without paging--c
        lock = {_M_device = 0x555555797578, _M_owns = false}
#9  std::_V2::condition_variable_any::wait_for<std::unique_lock<std::mutex>, long, std::ratio<1l, 1000l> >(std::unique_lock<std::mutex>&, std::chrono::duration<long, std::ratio<1l, 1000l> > const&) (__rtime=<optimized out>, __lock=..., this=0x5555557975a0) at /usr/include/c++/11/condition_variable:355
        lock = {_M_device = 0x555555797578, _M_owns = false}
#10 FreeDVInterface::EventHandlerThread<FreeDVInterface::RxAudioThreadState*, FreeDVInterface::RxAudioThreadState*>::ThreadEntry_(FreeDVInterface::EventHandlerThread<FreeDVInterface::RxAudioThreadState*, FreeDVInterface::RxAudioThreadState*>*) (ptr=0x555555797520) at /usr/src/debug/freedv-1.6.0-0.0.20.fc34.x86_64/src/freedv_interface.h:223
        lock = {_M_device = 0x555555797578, _M_owns = false}
#11 0x00007ffff5961c84 in std::execute_native_thread_routine(void*) (__p=0x555555e7bb90) at ../../../../../libstdc++-v3/src/c++11/thread.cc:82
#12 0x00007ffff5711299 in start_thread (arg=0x7fffc37fe640) at pthread_create.c:481
        ret = <optimized out>
        pd = 0x7fffc37fe640
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140736473327168, 7119152471315322813, 140737488337214, 140737488337215, 0, 140736473327168, -7119124984473577539, -7119164522035144771}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = 0
#13 0x00007ffff5639353 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 80 (Thread 0x7fffc3fff640 (LWP 227977) "freedv"):
#0  0x00007ffff571da8a in __futex_abstimed_wait_common64 (futex_word=futex_word@entry=0x5555557c3088, expected=expected@entry=0, clockid=clockid@entry=1, abstime=abstime@entry=0x7fffc3ffea70, private=private@entry=0, cancel=cancel@entry=true) at ../sysdeps/nptl/futex-internal.c:74
        sc_cancel_oldtype = 0
        sc_ret = <optimized out>
        clockbit = 0
        err = <optimized out>
        op = 137
#1  0x00007ffff571daef in __GI___futex_abstimed_wait_cancelable64 (futex_word=futex_word@entry=0x5555557c3088, expected=expected@entry=0, clockid=clockid@entry=1, abstime=abstime@entry=0x7fffc3ffea70, private=private@entry=0) at ../sysdeps/nptl/futex-internal.c:123
#2  0x00007ffff5717892 in __pthread_cond_wait_common (abstime=0x7fffc3ffea70, clockid=1, mutex=0x555555d401b0, cond=0x5555557c3060) at pthread_cond_wait.c:504
        spin = 0
        buffer = {__routine = 0x7ffff5717040 <__condvar_cleanup_waiting>, __arg = 0x7fffc3ffe990, __canceltype = -134440518, __prev = 0x0}
        cbuffer = {wseq = 0, cond = 0x5555557c3060, mutex = 0x555555d401b0, private = 0}
[... 693 lines cut ...]
        source = 0x5555556ee8a0
        current = 0x5555556d68c0
        i = 0
#56 g_main_context_dispatch (context=0x5555556d6800) at ../glib/gmain.c:4055
#57 0x00007ffff46444e8 in g_main_context_iterate.constprop.0 (context=0x5555556d6800, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/gmain.c:4131
        max_priority = 2147483647
        timeout = 367
        some_ready = 1
        nfds = 4
        allocated_nfds = 5
        fds = <optimized out>
        begin_time_nsec = 45171791579572
#58 0x00007ffff45efa93 in g_main_loop_run (loop=0x555555edf6c0) at ../glib/gmain.c:4329
        __func__ = "g_main_loop_run"
#59 0x00007ffff4c33e1d in gtk_main () at /usr/src/debug/gtk3-3.24.30-1.fc34.x86_64/gtk/gtkmain.c:1329
        loop = 0x555555edf6c0
#60 0x00007ffff62a1475 in wxGUIEventLoop::DoRun() (this=0x555555f45c20) at ../src/gtk/evtloop.cpp:65
        loopLevel = 0
#61 0x00007ffff5e8c5e1 in wxEventLoopBase::Run() (this=0x555555f45c20) at ../src/common/evtloopcmn.cpp:78
        activate = {m_evtLoopOld = 0x0}
#62 0x00007ffff5e66eda in wxAppConsoleBase::MainLoop() (this=0x5555556b15e0) at ../src/common/appbase.cpp:334
#63 0x00007ffff5ec86f0 in wxEntry(int&, wchar_t**) (argc=<optimized out>, argv=<optimized out>) at ../src/common/init.cpp:506
        initializer = {m_ok = <optimized out>}
#64 0x000055555559fa92 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /usr/src/debug/freedv-1.6.0-0.0.20.fc34.x86_64/src/main.cpp:181
(gdb)

Tyrbiter, 2021-08-20:
I used the build_linux.sh method and ran the resulting freedv binary, which seems to run OK, so maybe this is a build issue with my codec2 git checkout.

drowe67, 2021-08-20:
@Tyrbiter I'm wonder if if this is a side effect of the freedv-gui instability you have been encountering on your Fedora system, rather than a codec2 issue?

```
ofdm_set_verbose (ofdm=0x0, level=-1) at /usr/src/debug/codec2-1.0.0-1.fc34.2.x86_64/src/ofdm.c:971
971 void ofdm_set_verbose(struct OFDM *ofdm, int level) { ofdm->verbose = level; }
```
A verbosity `level=-1` suggests something upstream is feeding invalid verbosity values.

drowe67, 2021-08-20:
@Tyrbiter can you please run the codec2 ctests and see if they pass?

It would be useful to demonstrate that there is indeed an issue at the codec2 level, rather than when used in conjunction with freedv-gui, which already has some stability issues on your Fedora 34 system as per https://github.com/drowe67/freedv-gui/issues/159

Tyrbiter, 2021-08-21:
I can do this, but I don't know how to yet. Please could you let me know how to do it.

Tyrbiter, 2021-08-21:
I have been trying to resolve this but so far without success. Here is part of the thread stack frame that crashes from gdb:

Thread 1 "freedv" received signal SIGSEGV, Segmentation fault.
ofdm_set_verbose (ofdm=0x0, level=-1) at /usr/src/debug/codec2-1.0.0-1.fc34.2.x86_64/src/ofdm.c:971
971	void ofdm_set_verbose(struct OFDM *ofdm, int level) { ofdm->verbose = level; }
(gdb) bt full
#0  ofdm_set_verbose (ofdm=0x0, level=-1) at /usr/src/debug/codec2-1.0.0-1.fc34.2.x86_64/src/ofdm.c:971
#1  0x0000555555606873 in FreeDVInterface::setVerbose(bool) (this=<optimized out>, val=<optimized out>)
    at /usr/src/debug/freedv-1.6.1-0.0.1.fc34.x86_64/src/freedv_interface.cpp:379
        dv = @0x5555556aff50: 0x555555f94000
        __for_range = <optimized out>
#2  0x00005555555e1dfc in MainFrame::OnTogBtnOnOff(wxCommandEvent&) (this=0x555555868200, event=<optimized out>)
    at /usr/src/debug/freedv-1.6.1-0.0.1.fc34.x86_64/src/main.cpp:1610
        i = <optimized out>
        src_error = 7
        startStop = {static npos = 18446744073709551615, m_impl = L"&Start", m_convertedToChar = {m_str = 0x0, m_len = 4}}

The code from main.cpp:1610 is:

       // Codec2 verbosity setting
        freedvInterface.setVerbose(g_freedv_verbose);

then freedv_interface.cpp:379 which is (from a few lines before):

void FreeDVInterface::setVerbose(bool val)
{
    for (auto& dv : dvObjects_)
    {
        freedv_set_verbose(dv, val ? 2 : 0);
    }
}

So it looks as if the value passed, as you say, is -1, which gets set in main.cpp:527 and 528

    g_verbose = pConfig->Read(wxT("/Debug/verbose"), (long)0);
    g_freedv_verbose = pConfig->Read(wxT("/Debug/APIverbose"), (long)0);

Looking in the config file:

[bdm@deangelis SPECS]$ grep verbose ~/.freedv
verbose=0
APIverbose=0

On that basis I will confess I don't understand how the 0 in APIverbose can get turned into -1, unless there is a missing definition in main.h, where I find this:

extern int                 g_verbose;

but nothing for g_freedv_verbose which is defined in main.cpp

int                 g_freedv_verbose;

as is:

int                 g_verbose;

Could this be memory corruption? Variable being overwritten?

You may guess that I am not a programmer ;-)

tmiw, 2021-08-21:
Think we could get [libasan](https://github.com/google/sanitizers/wiki/AddressSanitizer) compiled in again so we can see where exactly it's corrupting memory? For reference, "-fsanitize=address -fno-omit-frame-pointer" needs to be added into freedv-gui/codec2's CFLAGS (along with possibly -lasan for LDFLAGS if using gcc instead of clang).

Tyrbiter, 2021-08-21:
Do you know where I need to add the -lasan? Looking at the various CMAKE files I can only find it relating to PORTAUDIO

Tyrbiter, 2021-08-21:
> Think we could get [libasan](https://github.com/google/sanitizers/wiki/AddressSanitizer) compiled in again so we can see where exactly it's corrupting memory? For reference, "-fsanitize=address [...]

Can you tell me where I need to add the -lasan? Can't find LDFLAGS in any of the obvious files in the codec2 source.

Tyrbiter, 2021-08-21:
Out of interest, if I use the options dialogue before starting the modem to set debug and APIdebug enabled, I then get exactly the same crash in ofdm.c:971 but with level=1 which I would expect to be a legal value.

drowe67, 2021-08-21:
> I can do this, but I don't know how to yet. Please could you let me know how to do it.

Instructions are in the codec2/README.md - you will need to work out which Fedora packages to install (the Ubuntu ones are listed).

drowe67, 2021-08-21:
> Out of interest, if I use the options dialogue before starting the modem to set debug and APIdebug enabled, I then get exactly the same crash in ofdm.c:971 but with level=1 which I would expect to be a legal value.

Taking a look at:
```
ofdm_set_verbose (ofdm=0x0, level=-1) at /usr/src/debug/codec2-1.0.0-1.fc34.2.x86_64/src/ofdm.c:971
971 void ofdm_set_verbose(struct OFDM *ofdm, int level) { ofdm->verbose = level; }
```
The first argument is a 0, but should be a memory address.  So it's getting passed a null pointer from further up.  This is looking more and more like a freeddv-gui problem (at least on your OS/hardware configuration).

drowe67, 2021-08-21:
@Tyrbiter could you pls:
1. Tell us what sort of hardware (machine and sound cards) you are using for this test?
1. Try the same configuration (Fedora 34), but on another machine if you have one available?
1. Try with completely different sound hardware.  Laptop sound cards are usually pretty solid. Start with a "Rx only" configuration (Tx sound cards set to None)

I've occasionally found that some sound card + PortAudio + Alsa leads to odd behavior.

Tyrbiter, 2021-08-21:
> > Out of interest, if I use the options dialogue before starting the modem to set debug and APIdebug enabled, I then get exactly the same crash in ofdm.c:971 but with level=1 which I would expect to be a legal value.
> 
> Taking a look at:
> 
> ```
> ofdm_set_verbose (ofdm=0x0, level=-1) at /usr/src/debug/codec2-1.0.0-1.fc34.2.x86_64/src/ofdm.c:971
> 971 void ofdm_set_verbose(struct OFDM *ofdm, int level) { ofdm->verbose = level; }
> ```
> 
> The first argument is a 0, but should be a memory address. So it's getting passed a null pointer from further up. This is looking more and more like a freedv-gui problem (at least on your OS/hardware configuration).

OK, I missed that, I assume then that the null pointer is from the function at freedv_interface.cpp:379, not sure how that passes arguments into codec2 ofdm.c, my code reading abilities are not that great.

Hardware is an Asus H370Mplus m/board with an Intel 9900 8-core 3GHz CPU, the sound cards are TI/Burr-Brown PCM2903C USB codecs plugged into a USB 3.0 hub PCI card (Renesas uP/USB ). OS is Fedora 34 with the latest updates and some self-built rpms where Richard Shaw @hobbes1069 has not yet released them to F34 updates (codec2, freedv, hamlib). Freedv is based on master with some recent ms-* branch changes carefully hacked into it, just the ms-ui-sideband and ms-pskreporter-improvements. All of these build and run OK, except for this recent problem. Building from the latest freedv-gui git master using build_linux.sh produces a freedv binary with codec2 and lpcnet included that runs OK.

I have also updated portaudio to use the April 2021 codebase, previously it dated from 2014, this version is in the rawhide distros so it will probably be in Fedora 35 in October.

I don't have Fedora on a laptop, it has Windows 10 on it and isn't multi-boot, my old Linux machine is hardly able to run the Fedora GUIs these days so I don't think I can easily try anything out on it.

Hope that isn't too unhelpful.

Tyrbiter, 2021-08-21:
> > I can do this, but I don't know how to yet. Please could you let me know how to do it.
> 
> Instructions are in the codec2/README.md - you will need to work out which Fedora packages to install (the Ubuntu ones are listed).

The part where rm -Rf build_linux happens breaks this, not sure how to fix it as the codec2 git tree doesn't have that directory.

tmiw, 2021-08-22:
> > Think we could get [libasan](https://github.com/google/sanitizers/wiki/AddressSanitizer) compiled in again so we can see where exactly it's corrupting memory? For reference, "-fsanitize=address -fno-omit-frame-pointer" needs to be added into freedv-gui/codec2's CFLAGS (along with possibly -lasan for LDFLAGS if using gcc instead of clang).
> 
> CFLAGS bit done.
> 
> Can you tell me where I need to add the -lasan? Can't find LDFLAGS in any of the obvious files in the codec2 source.

The hacky way of doing it is editing build_linux/CMakeCache.txt and changing the following:

```
CMAKE_EXE_LINKER_FLAGS:STRING=
```

to

```
CMAKE_EXE_LINKER_FLAGS:STRING=-lasan
```

followed by running "make" again. Otherwise you can add this somewhere in CMakeLists.txt and run "make" again:

```
set(CMAKE_EXE_LINKER_FLAGS "-lasan")
```

Tyrbiter, 2021-08-22:
OK, now have that done for the linker, but the gdb output is the same as before, do I need to do the same for freedv? I currently don't have the asan stuff compiled in.

Tyrbiter, 2021-08-22:
@hobbes1069 has just updated codec2, freedv and lpcnetfreedv packages for Fedora 34. I have downloaded and installed them and the crash in ofdm.c is now gone. I can now dissect the src rpms and work out why my builds did not work properly. As Richard is an official Fedora packager I can learn from his packaging nous.

If @drowe67 and @tmiw are in agreement then I think it's sensible to close this issue. Thanks for your help, it has been very instructive even if the problem is frustrating.

tmiw, 2021-08-22:
@Tyrbiter, no worries, glad that everything worked out in the end 👍 

@drowe67, I wonder if some stuff should be moved around inside LPCNet and Codec2 so that there's no circular dependency issues? It'd make build_linux.sh, etc. run faster if nothing else.

drowe67, 2021-08-22:
> @drowe67, I wonder if some stuff should be moved around inside LPCNet and Codec2 so that there's no circular dependency issues? It'd make build_linux.sh, etc. run faster if nothing else.

I have some ideas on that - it's actually part of https://github.com/drowe67/codec2/discussions/219 so will add a note there.

closed 2021-08-22 by drowe67