drowe67/codec2-dev #255: codec2 does not build on MSYS2/MinGW with GCC 11.2: file alloca.h not found

LigH-de, 2021-11-07:
```
FAILED: src/CMakeFiles/fsk_get_test_bits.dir/fsk_get_test_bits.c.o 
F:\MABS\msys64\mingw32\bin\ccache.exe  gcc -DGIT_HASH=\"v1.0.0-230-ga801aed3\" -IF:/MABS/build/codec2-git/build-32bit -IF:/MABS/build/codec2-git/src -mthreads -mtune=generic -O2 -pipe -Wall -Wno-strict-overflow -std=gnu11 -O3 -MD -MT src/CMakeFiles/fsk_get_test_bits.dir/fsk_get_test_bits.c.o -MF src\CMakeFiles\fsk_get_test_bits.dir\fsk_get_test_bits.c.o.d -o src/CMakeFiles/fsk_get_test_bits.dir/fsk_get_test_bits.c.o -c F:/MABS/build/codec2-git/src/fsk_get_test_bits.c
F:/MABS/build/codec2-git/src/fsk_get_test_bits.c:32:10: fatal error: alloca.h: No such file or directory
   32 | #include <alloca.h>
      |          ^~~~~~~~~~
compilation terminated.
```

srsampson, 2021-11-07:
A quick fix is to comment out the ```#include <alloca.h>``` and change:
https://github.com/drowe67/codec2/blob/master/src/fsk_get_test_bits.c#L77

to ```malloc()```

drowe67, 2021-11-07:
@srsampson that seems to be a side effect of https://github.com/drowe67/codec2/pull/253. 

A couple of thoughts:
1. What's the fix going forward?
1. Why isn't `alloca.h` included in this version of gcc?
1. Can we automatically test builds on this platform with GitHub actions?

srsampson, 2021-11-07:
I recommend getting rid of alloc() in that file. It's useless, when malloc() can do the job just as well, and is portable to DOS 3.1.

srsampson, 2021-11-07:
> ...that seems to be a side effect of #253.
> 

Yes, It looks like the DOS programs use malloc.h which on their systems has a link to _alloc().
```
#ifdef __GNUC__
#undef _alloca
#define _alloca(x) __builtin_alloca((x))
#else
  void *__cdecl _alloca(size_t _Size) __MINGW_ATTRIB_DEPRECATED_SEC_WARN;
#endif
```

So I guess for Linux we need alloc() but for DOS we need malloc().

Maybe we should get rid of alloc() in that program. Is there some special need for the data to be on the stack versus the heap?

tmiw, 2021-11-07:
There might be value in having it on the stack if it was going to be run on a memory-constrained system (e.g. DOS without HIMEM.SYS loaded) but that seems unlikely considering the purpose of fsk_get_test_bits. I'd say converting to malloc is good (or even simply moving the definition of bitbuf down to that line and replacing it with an array definition if we wanted to be 100% safe).

drowe67, 2021-11-08:
Yes I like the idea of removing the `alloca()` calls, and either using `malloc()` or variable length arrays.

srsampson, 2021-11-08:
> Yes I like the idea of removing the `alloca()` calls, and either using `malloc()` or variable length arrays.

I was going to use an array, as it's only 100 bytes or so. Guess I wanted to stay with his original spirit of design, ha.

closed 2021-11-11 by drowe67