drowe67/codec2-dev #257: record the sound in raw format

mahdishakeri88, 2021-11-09:
Hi dears
I want to record the sound and use the Codec 2 to compress it then save it in memory and decompress it again, 
but the format that the program accepts is raw. How can I record the sound in this format?
Also, the audio files inside the codec 2 package have a good quality after decompressing, but when I give the file being recorded to the program and compress and then decompress it (according to the following command), its quality drops sharply. 
br=1300; arecord -f S16_LE -c 1 -r 8000 | ./src/c2enc $br - - | ./src/c2dec $br - - | aplay -f S16_LE -
What is the problem ?
Tnx to all

Posejdos, 2021-11-09:
Well, if you can hear the output and understand it, then you're probably recording it right. Try saving the recorded file on your disc and playing it, so you have a reference point depending on quality of your microphone and speakers.

Possible this is just how the codec2 performs in that bitrate.

tmiw, 2021-11-09:
Can you try processing one of the sample files? For example, https://github.com/drowe67/codec2/blob/master/raw/ve9qrp_10s.raw. That should rule out any potential configuration issues on your end.

closed 2021-11-11 by drowe67