drowe67/codec2-dev #268: MinGW Codec2 build failures

tmiw, 2021-12-25:
Just tried to generate a freedv-gui Windows build and ended up with the following failures:

```
fdv_win_fed34_c     | --------------------- starting build_windows.sh ---------------------
fdv_win_fed34_c     | Cloning into 'codec2'...
fdv_win_fed34_c     | Already on 'master'
fdv_win_fed34_c     | Your branch is up to date with 'origin/master'.
fdv_win_fed34_c     | Already up to date.
fdv_win_fed34_c     | -- The C compiler identification is GNU 10.3.1
fdv_win_fed34_c     | -- The CXX compiler identification is GNU 10.3.1
fdv_win_fed34_c     | -- Detecting C compiler ABI info
fdv_win_fed34_c     | -- Detecting C compiler ABI info - done
fdv_win_fed34_c     | -- Check for working C compiler: /usr/bin/x86_64-w64-mingw32-gcc - skipped
fdv_win_fed34_c     | -- Detecting C compile features
fdv_win_fed34_c     | -- Detecting C compile features - done
fdv_win_fed34_c     | -- Detecting CXX compiler ABI info
fdv_win_fed34_c     | -- Detecting CXX compiler ABI info - done
fdv_win_fed34_c     | -- Check for working CXX compiler: /usr/bin/x86_64-w64-mingw32-g++ - skipped
fdv_win_fed34_c     | -- Detecting CXX compile features
fdv_win_fed34_c     | -- Detecting CXX compile features - done
fdv_win_fed34_c     | -- codec2 version: 1.0
fdv_win_fed34_c     | -- freedv-gui current git hash: v1.0.0-321-gbc8b049b
fdv_win_fed34_c     | -- Performing Test COMPILER_SUPPORTS_GNU11
fdv_win_fed34_c     | -- Performing Test COMPILER_SUPPORTS_GNU11 - Success
fdv_win_fed34_c     | -- Performing Test COMPILER_SUPPORTS_GNU99
fdv_win_fed34_c     | -- Performing Test COMPILER_SUPPORTS_GNU99 - Success
fdv_win_fed34_c     | -- System is MinGW.
fdv_win_fed34_c     | -- Found Git: /usr/bin/git (found version "2.31.1") 
fdv_win_fed34_c     | -- Codec2 current git hash: v1.0.0-321-gbc8b049b
fdv_win_fed34_c     | -- Looking for include file stdlib.h
fdv_win_fed34_c     | -- Looking for include file stdlib.h - found
fdv_win_fed34_c     | -- Looking for include file string.h
fdv_win_fed34_c     | -- Looking for include file string.h - found
fdv_win_fed34_c     | -- Looking for __GNU_LIBRARY__
fdv_win_fed34_c     | -- Looking for __GNU_LIBRARY__ - not found
fdv_win_fed34_c     | -- Looking for _GNU_SOURCE
fdv_win_fed34_c     | -- Looking for _GNU_SOURCE - not found
fdv_win_fed34_c     | -- Looking for floor
fdv_win_fed34_c     | -- Looking for floor - found
fdv_win_fed34_c     | -- Looking for ceil
fdv_win_fed34_c     | -- Looking for ceil - found
fdv_win_fed34_c     | -- Looking for pow
fdv_win_fed34_c     | -- Looking for pow - found
fdv_win_fed34_c     | -- Looking for sqrt
fdv_win_fed34_c     | -- Looking for sqrt - found
fdv_win_fed34_c     | -- Looking for sin
fdv_win_fed34_c     | -- Looking for sin - found
fdv_win_fed34_c     | -- Looking for cos
fdv_win_fed34_c     | -- Looking for cos - found
fdv_win_fed34_c     | -- Looking for atan2
fdv_win_fed34_c     | -- Looking for atan2 - found
fdv_win_fed34_c     | -- Looking for log10
fdv_win_fed34_c     | -- Looking for log10 - found
fdv_win_fed34_c     | -- Looking for round
fdv_win_fed34_c     | -- Looking for round - found
fdv_win_fed34_c     | -- Looking for getopt
fdv_win_fed34_c     | -- Looking for getopt - found
fdv_win_fed34_c     | -- Build type is: Debug
fdv_win_fed34_c     | -- Compiler Flags:  -Wall -Wno-strict-overflow -std=gnu11-g -O2 -DDUMP
fdv_win_fed34_c     | -- Libraries linked: 
fdv_win_fed34_c     | -- Compilation date = XX20211225XX
fdv_win_fed34_c     | -- Configuring done
fdv_win_fed34_c     | -- Generating done
fdv_win_fed34_c     | CMake Warning:
fdv_win_fed34_c     |   Manually-specified variables were not used by the project:
fdv_win_fed34_c     | 
fdv_win_fed34_c     |     INCLUDE_INSTALL_DIR
fdv_win_fed34_c     |     SHARE_INSTALL_PREFIX
fdv_win_fed34_c     |     SYSCONF_INSTALL_DIR
fdv_win_fed34_c     | 
fdv_win_fed34_c     | 
fdv_win_fed34_c     | -- Build files have been written to: /home/build/freedv-gui/codec2/build_win64
fdv_win_fed34_c     | Scanning dependencies of target codec2_native
fdv_win_fed34_c     | [  0%] Creating directories for 'codec2_native'
fdv_win_fed34_c     | [  1%] No download step for 'codec2_native'
fdv_win_fed34_c     | [  1%] No update step for 'codec2_native'
fdv_win_fed34_c     | [  2%] No patch step for 'codec2_native'
fdv_win_fed34_c     | [  2%] Performing configure step for 'codec2_native'
fdv_win_fed34_c     | -- The C compiler identification is GNU 11.0.1
fdv_win_fed34_c     | -- The CXX compiler identification is unknown
fdv_win_fed34_c     | -- Detecting C compiler ABI info
fdv_win_fed34_c     | -- Detecting C compiler ABI info - done
fdv_win_fed34_c     | -- Check for working C compiler: /usr/bin/cc - skipped
fdv_win_fed34_c     | -- Detecting C compile features
fdv_win_fed34_c     | -- Detecting C compile features - done
fdv_win_fed34_c     | CMake Error at CMakeLists.txt:10 (project):
fdv_win_fed34_c     |   No CMAKE_CXX_COMPILER could be found.
fdv_win_fed34_c     | 
fdv_win_fed34_c     |   Tell CMake where to find the compiler by setting either the environment
fdv_win_fed34_c     |   variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
fdv_win_fed34_c     |   to the compiler, or to the compiler name if it is in the PATH.
fdv_win_fed34_c     | 
fdv_win_fed34_c     | 
fdv_win_fed34_c     | -- Configuring incomplete, errors occurred!
```

Any idea what could be going on?

hobbes1069, 2021-12-25:
I know I've been out of the loop a bit. What are the system details? Is this in some sort of CI environment?

The error is that it can't find the native (not the MinGW) c++ compiler, I assume it's installed? If I remember correctly the only reason we need this is for codebook generation.

tmiw, 2021-12-25:
> I know I've been out of the loop a bit. What are the system details? Is this in some sort of CI environment?
> 
> The error is that it can't find the native (not the MinGW) c++ compiler, I assume it's installed? If I remember correctly the only reason we need this is for codebook generation.

It's a Docker container set up to compile FreeDV, codec2 and LPCNet for Windows. The files for that are [here](https://github.com/drowe67/freedv-gui/tree/master/docker). I would have thought that the following would have set up the native g++ as well, no?

```
dnf -y install --setopt=install_weak_deps=False @development-tools cmake git speexdsp-devel libsamplerate-devel \
    octave octave-signal gnuplot sox python3-numpy automake libtool libusb1-devel wget bc glibc.i686 which \ 
    bzip2 ncurses-compat-libs && useradd -m build 
```

hobbes1069, 2021-12-25:
Hah, I wasn't aware of the @ trick to install a group, but that does make it a one liner instead of having to install it separately.

That being said, it doesn't look like gcc or gcc-c++ are a direct dependency of the group so something is pulling in gcc, but not gcc-c++. Even though gcc is being pulled in by something I wouldn't rely on that so I would add "gcc gcc-c++" to the dnf options.
```
# dnf group info "Development Tools"
Last metadata expiration check: 2:37:10 ago on Sat 25 Dec 2021 08:06:37 AM CST.
Group: Development Tools
 Description: These tools include general development tools such as git and CVS.
 Mandatory Packages:
   gettext
 Default Packages:
   diffstat
   doxygen
   git
   patch
   patchutils
   subversion
   systemtap
 Optional Packages:
   buildbot
   colordiff
   cvs
   cvs2cl
   cvsgraph
   cvsps
   darcs
   dejagnu
   expect
   gambas3-ide
   git-annex
   git-cola
   git2cl
   gitg
   gtranslator
   highlight
   lcov
   manedit
   meld
   monotone
   myrepos
   nemiver
   qgit
   quilt
   rapidsvn
   rcs
   robodoc
   scanmem
   subunit
   svn2cl
   tig
   tkcvs
   tortoisehg
   translate-toolkit
   utrac
```

tmiw, 2021-12-25:
Yep, looks like adding gcc and gcc-c++ fixed the issue. This probably never came up before because we were forcing the project to be C only before #267, which made Codec2 a dual C and C++ project to enable CML to be properly built. Thanks for the help and have a good holiday! 👍

closed 2021-12-25 by tmiw