drowe67/codec2-dev #288: cannot compile for microcontroller

jerabaul29, 2022-01-28:
I am trying to compile the library for microcontroller use.  I am trying to use the CMakeLists.txt for microcontrollers that is recommended in the Readme, but I am facing problems. Not sure if I am misunderstanding something or doing something wrong or else.

I first have to apply a couple of changes to get cmake to run without issues; after a bit of googling, I end up with:

```
cmake_minimum_required(VERSION 3.16)

set(CMAKE_C_COMPILER "/home/jrmet/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gcc")
set(CMAKE_CXX_COMPILER "/home/jrmet/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-c++")

set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
project(codec2staticlib)
set(MICROCONTROLLER_BUILD 1)

set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mlittle-endian -ffunction-sections -fdata-sections -g -O3")
set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -ffunction-sections -fdata-sections")

add_definitions(-DCORTEX_M4 -D__EMBEDDED__)
add_definitions(-DFREEDV_MODE_EN_DEFAULT=0 -DFREEDV_MODE_1600_EN=1 -DFREEDV_MODE_700D_EN=1 -DFREEDV_MODE_700E_EN=1 -DCODEC2_MODE_EN_DEFAULT=0 -DCODEC2_MODE_1300_EN=1 -DCODEC2_MODE_700C_EN=1)
                                        
include(FetchContent)

FetchContent_Declare(codec2
    GIT_REPOSITORY https://github.com/drowe67/codec2.git
    GIT_TAG origin/master
    GIT_SHALLOW ON
    GIT_PROGRESS ON
)
FetchContent_GetProperties(codec2)
if(NOT ${codec2_POPULATED})
    FetchContent_Populate(codec2)
endif()
set(CMAKE_REQUIRED_FLAGS "")

set(LPCNET OFF CACHE BOOL "")
add_subdirectory(${codec2_SOURCE_DIR} ${codec2_BINARY_DIR} EXCLUDE_FROM_ALL)

set(CMAKE_SHARED_LIBRARY_LINK_C_FLAGS "")
set(CMAKE_SHARED_LIBRARY_LINK_CXX_FLAGS "")
```

this then lets me do:

```
rm -Rf build_linux && mkdir build_linux
cd build_linux/
```

which runs fine.

However, doing:

```
make all
```

does nothing, and doing:

```
make codec2
```

produces:

```
~/Desktop/Git/codec2/build_linux > make codec2
Scanning dependencies of target generate_codebook
[  2%] Building C object _deps/codec2-build/src/CMakeFiles/generate_codebook.dir/generate_codebook.c.o
[  2%] Linking C executable generate_codebook
arm-none-eabi-gcc: error: unrecognized command line option '-rdynamic'
make[3]: *** [_deps/codec2-build/src/CMakeFiles/generate_codebook.dir/build.make:84: _deps/codec2-build/src/generate_codebook] Error 1
make[2]: *** [CMakeFiles/Makefile2:826: _deps/codec2-build/src/CMakeFiles/generate_codebook.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:860: _deps/codec2-build/src/CMakeFiles/codec2.dir/rule] Error 2
make: *** [Makefile:485: codec2] Error 2
```

Any idea what I am doing wrong? All I want is to generate a static library of codec2 so that I can decode / encode on an arm cortex m micro controller.

drowe67, 2022-01-28:
The `codec2/stm32` build system is another possible starting point.  Also see https://github.com/drowe67/codec2/discussions/286

jerabaul29, 2022-01-28:
Thanks for the link - interesting somebody was asking something so similar recently :) .

I am a bit in doubt about what the best way to proceed is because the project is developed in PlatformIO that comes with its own build system. Will have to think a bit about it 😊😅