drowe67/codec2-dev #293: Unable to build with -Os

ZeroChaos-, 2022-01-31:
I realize -O2 or -O3 are probably a lot more common, but the code breaks when build is attempted with -Os on gcc

```
[138/201] : && /usr/bin/x86_64-pc-linux-gnu-gcc -m32 -Os -march=native -mtune=native -pipe -frecord-gcc-switches -Wall -Wno-strict-overflow -std=gnu11 -fPIC -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0    -rdynamic src/CMakeFiles/ldpc_dec.dir/ldpc_dec.c.o -o src/ldpc_dec  -Wl,-rpath,/var/tmp/portage/media-libs/codec2-1.0.3/work/codec2-1.03_build-abi_x86_32.x86/src  -lm  src/libcodec2.so.1.0  -lm && :
FAILED: src/ldpc_dec 
: && /usr/bin/x86_64-pc-linux-gnu-gcc -m32 -Os -march=native -mtune=native -pipe -frecord-gcc-switches -Wall -Wno-strict-overflow -std=gnu11 -fPIC -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0    -rdynamic src/CMakeFiles/ldpc_dec.dir/ldpc_dec.c.o -o src/ldpc_dec  -Wl,-rpath,/var/tmp/portage/media-libs/codec2-1.0.3/work/codec2-1.03_build-abi_x86_32.x86/src  -lm  src/libcodec2.so.1.0  -lm && :
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: src/libcodec2.so.1.0: undefined reference to `rx_filter_coh'
collect2: error: ld returned 1 exit status
```

I suspect this is related to the failed https://github.com/drowe67/codec2/pull/240

tmiw, 2022-01-31:
I wonder if this is some sort of compiler bug. Even with -Os I would expect `rx_filter_coh()` not to get fully stripped out as it's obviously being referenced by another function.

Regardless, I'm fine with removing "inline" for it. @drowe67, does that sound reasonable?

hogthrob, 2022-01-31:
Hi,
this is not strange but documented behavior of a C compiler in C99 mode,

see https://stackoverflow.com/questions/216510/what-does-extern-inline-do

So fix is simple: put `extern` before `inline` and all is good. Code is inlined and also provided as linkable translation unit.

drowe67, 2022-01-31:
Thanks @ZeroChaos- @tmiw @hogthrob

closed 2022-01-31 by drowe67