drowe67/codec2-dev #305: 1700 Hz carrier syncs as 700C

tmiw, 2022-02-14:
Noticed this while operating portable today. Someone was transmitting a constant carrier around 1700 Hz in my passband and FreeDV 1.8.0 (using codec2 from the dr-2020A branch) seemed to falsely sync to that signal using 700C. Not sure if that's anything that can be mitigated but figured I'd create an issue about it anyway just in case. 

(2020A/B were not active at the time, BTW, so it's unlikely any changes for those would cause this.)

drowe67, 2022-02-14:
It might be interesting to try ` ./misc/sine - 1700 60 | ./freedv_rx 700C /dev/null --vv` and see if 700C syncs up.

tmiw, 2022-02-14:
See below:

```
HT-TM05:src mooneer$ ./sine - 1700 60 | ./freedv_rx --vv 700C - /dev/null
  [0] acohpsk.f_est: 1460.000000 +/- 20
  [0]   fine freq f:  -7.25 max_ratio: 0.928958 ct: 5
  [0]   encouraging sync word!
  [0] acohpsk.f_est: 1500.000000 +/- 20
  [0]   fine freq f: -16.50 max_ratio: 0.961555 ct: 5
  [0]   encouraging sync word!
  [0] acohpsk.f_est: 1540.000000 +/- 20
  [0]   fine freq f: -18.50 max_ratio: 0.964278 ct: 5
  [0]   encouraging sync word!
  [0] trying sync and f_est: 1558.500000
  [0]   fine freq f: -20.00 max_ratio: 0.942351 ct: 5
  [0]   encouraging sync word!
  [0] Hmm -20.000000 is a bit big :(
  [0] acohpsk.f_est: 1460.000000 +/- 20
  [0]   fine freq f: -14.75 max_ratio: 0.995181 ct: 2
  [0]   encouraging sync word!
  [0] acohpsk.f_est: 1500.000000 +/- 20
  [0]   fine freq f:  19.25 max_ratio: 0.975804 ct: 1
  [0]   encouraging sync word!
  [0] acohpsk.f_est: 1540.000000 +/- 20
  [0]   fine freq f:  16.50 max_ratio: 0.975756 ct: 1
  [0]   encouraging sync word!
  [0] trying sync and f_est: 1474.750000
  [0]   fine freq f:   8.50 max_ratio: 0.996605 ct: 2
  [0]   encouraging sync word!
  [0] Hmm 8.500000 is a bit big :(
  [0] acohpsk.f_est: 1460.000000 +/- 20
  [0]   fine freq f:  -9.75 max_ratio: 0.978169 ct: 2
  [0]   encouraging sync word!
  [0] acohpsk.f_est: 1500.000000 +/- 20
  [0]   fine freq f:  19.00 max_ratio: 0.962151 ct: 5
  [0]   encouraging sync word!
  [0] acohpsk.f_est: 1540.000000 +/- 20
  [0]   fine freq f:  14.25 max_ratio: 0.969928 ct: 5
  [0]   encouraging sync word!
  [0] trying sync and f_est: 1469.750000
  [0]   fine freq f:   0.50 max_ratio: 0.995045 ct: 2
  [0]   encouraging sync word!
  [0] in sync! f_est: 1469.750000 ratio: 0.995045 
frames decoded: 750  output speech samples: 480000
HT-TM05:src mooneer$ 
```

Tyrbiter, 2022-03-05:
https://github.com/drowe67/freedv-gui/discussions/218

Tyrbiter, 2023-06-08:
Any further thoughts on this issue?

I'm hearing a lot of distorted decoded audio where a quick look at the radio's IF spectrum display or the freedv spectrum graph shows a fairly strong single tone in the passband. It would be nice if this could be looked at as it would make usability much improved, I can't see us persuading stations not to tune up :)

drowe67, 2023-06-09:
@Tyrbiter can you provide a sample please? It would be interesting to explore the issue as it happens.

As background I've tested modes with FEC (like 700D/E) and they are fairly robust to this.  700C is going to be hard to fix, as it doesn't have FEC.

Moving forward to new modes, we can set a spec for co-channel interference and develop against it.

Tyrbiter, 2023-06-11:
I will attach a recording once I get one, of course the problem has reduced incidence now that I've asked the question.

Good point about co-channel effects. Was that considered in the original development?

drowe67, 2023-06-21:
Thanks @Tyrbiter - good to have an example of this issue.

Tyrbiter, 2023-07-30:
More unusual example of 700E false syncing with codec2 v1.2.0 release. Usually see this with 700C.

[FreeDV_FromRadio_700E_false_sync_20230730-183308.zip](https://github.com/drowe67/codec2-dev/files/12209780/FreeDV_FromRadio_700E_false_sync_20230730-183308.zip)

drowe67, 2023-07-30:
Thanks @Tyrbiter - we'll use these examples to design tests for the new waveform.

Tyrbiter, 2023-07-31:
I will post any more of these that I manage to catch.

drowe67, 2023-08-05:
Thanks @Tyrbiter - but we probably have enough samples to characterise the problem and test the new waveform against this scenario.

Tyrbiter, 2023-08-05:
Oh, yeah, I meant that I would only send something unusual and new. The 2 or 3 provided have a single tone somewhere in the audio spectrum which I can see on the waterfall and so were easy to capture.

I hear some short nonsense decodes quite often which the modem sees as 2020 modes, capturing these is very difficult so I have pretty much given up on that. The modem appears to sync on noise or possibly on weak SSB signals, these cases are much less deterministic than the 700C and 700E cases.

Don't worry, I'm not going to bury you under .wav files :)