drowe67/codec2-dev #322: SM1000 build failure on Ubuntu 21.10

tmiw, 2022-05-16:
While working on #317 I tried to compile the SM1000 firmware on Ubuntu 21.10. Unfortunately I got the following:

```
[ 68%] Building C object CMakeFiles/codec2_prof.dir/codebooknewamp1_energy.c.obj
[ 68%] Building C object CMakeFiles/codec2_prof.dir/codebooknewamp1.c.obj
[ 68%] Linking C static library libcodec2_prof.a
[ 69%] Built target codec2_prof
Scanning dependencies of target sm1000base
[ 69%] Building C object CMakeFiles/sm1000base.dir/src/sm1000_leds_switches.c.obj
[ 70%] Building C object CMakeFiles/sm1000base.dir/src/debugblinky.c.obj
[ 70%] Building C object CMakeFiles/sm1000base.dir/src/system_stm32f4xx.c.obj
[ 70%] Building ASM object CMakeFiles/sm1000base.dir/src/startup_stm32f4xx.s.obj
[ 70%] Linking C static library libsm1000base.a
[ 70%] Built target sm1000base
[ 71%] Building C object CMakeFiles/dac_ut.dir/src/dac_ut.c.obj
[ 71%] Linking C executable dac_ut.elf
Creating binary for dac_ut
/usr/bin/objcopy: Unable to recognise the format of the input file `dac_ut.elf'
make[2]: *** [CMakeFiles/dac_ut.dir/build.make:102: dac_ut.elf] Error 1
make[2]: *** Deleting file 'dac_ut.elf'
make[1]: *** [CMakeFiles/Makefile2:1084: CMakeFiles/dac_ut.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```

Anyway, it seems that it's using the x86_64 version of objcopy instead of `arm-none-eabi-objcopy` as expected.

drowe67, 2022-07-28:
Hi @tmiw - can we close this one now?

tmiw, 2022-07-29:
Yep!

closed 2022-07-29 by tmiw