drowe67/codec2-dev #324: cross-compile using host native generate_codebook fails

th0ma7, 2022-05-19:
I am on my way to integrate `codec2` to SynoCommunity `ffmpeg` and I've faced an issue where it fails when cross-compiling on non-x86 platforms (e.g. armv5, v7, v8 and ppc) due to `/bin/sh: 1: ./generate_codebook: Exec format error`.

Reading the `READM.md` it refers to a `-DGENERATE_CODEBOOK` option that can be passed.  As such I'm pre-compiling a host native version of `codec2` including the `generate_codebook` binary that I can refer to when cross-compiling later on.  But this fails as it doesn't refer to it during the build but rather continue using the target cross-compiled version.

The `build/CMakeCache.txt` declare it as unititialized:
```
build/CMakeCache.txt:GENERATE_CODEBOOK:UNINITIALIZED=/home/spksrc/ffmpeg-extras/spksrc/cross/codec2/work-armv7-6.1/../../../native/codec2/work-native/install/usr/local/bin/generate_codebook
```

I've briefly looked in the code but didn't found any easy ways to circumvent this.  Help much apprecited, thnx in advance.

hobbes1069, 2022-05-19:
I'll have to take a look. It's been a few years since I wrote the code for that and was originally intended to allow me to build the codebook on linux while cross-compiling for Windows. Can you post a link to a full log somewhere?

th0ma7, 2022-05-19:
Sure, here a build log for armv7.  You'll notice the native build first followed by the cross-compiled that fails.

[build-armv7-6.1.log](https://github.com/drowe67/codec2/files/8732458/build-armv7-6.1.log)

tmiw, 2022-05-19:
Perhaps copying the strategy that the SM1000 build does will help? Per [stm32/CMakeLists.txt](https://github.com/drowe67/codec2/blob/master/stm32/CMakeLists.txt) it includes [stm32/cmake/gencodebooks.cmake](https://github.com/drowe67/codec2/blob/master/stm32/cmake/gencodebooks.cmake), which adds a new external project called `codec2_native` and some custom commands to generate the codebooks.

OTOH, I don't seem to need to do that for [my ezDV project](https://github.com/tmiw/ezDV/blob/main/firmware/main/CMakeLists.txt) to crosscompile for ESP32. That might be because ESP-IDF passes in a custom toolchain file when it executes CMake.

drowe67, 2022-05-19:
Until recently we actually had the pre-compiled codebooks sitting in the repo, I removed them in #285.  They rarely change, so we could put them back in I guess, to simplify cross compiling.

hobbes1069, 2022-05-19:
If we can come up with a more generalized solution pretty quickly I think that would be the optimal solution, but obviously not the quickest.

drowe67, 2022-05-19:
Yes I removed them as it felt like we were committing `.o` files.

tmiw, 2022-05-21:
@th0ma7, try this.

1. Download [arm-linux-toolchain.cmake.zip](https://github.com/drowe67/codec2/files/8747643/arm-linux-toolchain.cmake.zip). Uncompress it an edit CMAKE_SYSROOT and CMAKE_STAGING_PREFIX as appropriate.
2. Run the following commands:

```
sudo apt install gcc-11-aarch64-linux-gnu g++-11-aarch64-linux-gnu
git clone https://github.com/drowe67/codec2.git codec2-test
cd codec2-test
cp ~/arm-linux-toolchain.cmake .
mkdir build
cd build
cmake -DCMAKE_TOOLCHAIN_FILE=../arm-linux-toolchain.cmake ..
make
```

For me, this builds a 64 bit ARM Linux version with no problem on Ubuntu 21.10. You should be able to adjust the .cmake file and required packages to build 32 bit ARM instead if preferred.

th0ma7, 2022-05-21:
Integrating this into the atuomated build systems isn't that obvious.  Although I've added the `SYSROOT` and `STAGING_PREFIX`:
```
CMAKE_ARGS  = -DUNITTEST=FALSE
CMAKE_ARGS += -DCMAKE_STAGING_PREFIX=$(STAGING_INSTALL_PREFIX)
CMAKE_ARGS += -DCMAKE_SYSROOT=/home/spksrc/ffmpeg-codec2/spksrc/toolchain/syno-armv7-6.1/work/arm-unknown-linux-gnueabi/arm-unknown-linux-gnueabi/sysroot
CMAKE_ARGS += -DGENERATE_CODEBOOK=$(WORK_DIR)/../../../native/codec2/work-native/install/usr/local/bin/generate_codebook
```
But this leads to the same error `/bin/sh: 1: ./generate_codebook: Exec format error` as it is still built for arm target:
```
spksrc@spksrc:~/ffmpeg-codec2/spksrc/cross/codec2$ file ./work-armv7-6.1/codec2-1.0.3/build/src/generate_codebook
./work-armv7-6.1/codec2-1.0.3/build/src/generate_codebook: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, with debug_info, not stripped
```

tmiw, 2022-05-22:
> Integrating this into the atuomated build systems isn't that obvious. Although I've added the `SYSROOT` and `STAGING_PREFIX`:
>
> ```
> CMAKE_ARGS  = -DUNITTEST=FALSE
> CMAKE_ARGS += [...]

My understanding is that `-DCMAKE_TOOLCHAIN_FILE=...` has to be passed in as part of the `cmake` command for your project (or set inside that project's CMakeLists.txt) in order for the settings inside it to apply to codec2. More info [here](https://cmake.org/cmake/help/book/mastering-cmake/chapter/Cross%20Compiling%20With%20CMake.html) in CMake's documentation.

th0ma7, 2022-06-02:
@tmiw After a lot of readings and tuning of our build framework I've switched our CMake portion to using toolchain files instead of pointing to cross-toolchain thus hiding host native tools from cmake.  This in turn solved the issue.  Thnx.

For reference: https://github.com/SynoCommunity/spksrc/pull/5280

closed 2022-06-02 by th0ma7