drowe67/codec2-dev #335: 1.0.4 release confusion

Xiretza, 2022-07-15:
Hi! It looks like a broken `v1.0.4` tag was pushed a couple days ago, then marked as deprecated. Now there's a `v1.0.4_rc2` tag, which is confusing - shouldn't this be `v1.0.5_rc2`? There already is a 1.0.4 tag, calling the next (no longer broken) version 1.0.4 as well will create all kinds of conflicts.

drowe67, 2022-07-15:
Yes that's a good point.  @tmiw - do you agree I should do a 1.0.5 tag/release?

tmiw, 2022-07-15:
> Yes that's a good point. @tmiw - do you agree I should do a 1.0.5 tag/release? Hmm, would that break your https://github.com/drowe67/freedv-gui/releases/tag/v1.8.1 release?
> 
> I guess 1.0.5 could just be a tag/release at the same commit ....

CMakeLists.txt would need to be updated to be 1.0.5 so the 1.0.5 tag won't be able to be on the same commit. Regardless, as long as we keep the existing 1.0.4_rc2 tag, that would be fine, right?

drowe67, 2022-07-16:
https://github.com/drowe67/codec2/releases/tag/v1.0.5

closed 2022-07-16 by drowe67