drowe67/codec2-dev #345: compile error in an Android project

smilingacan, 2022-08-08:
hello，I met an error when including Codec 2 in an Android project。
The errors are as follows:
  [46/202] Performing configure step for 'codec2_native'
  -- Codec2 current git hash: 01313d26
  -- System is MinGW.
  -- codec2 version: 1.0.5
  -- Build type is: Debug
  -- Compiler Flags:  -Wall -Wno-strict-overflow -std=gnu11-g -O2 -DDUMP
  -- Libraries linked: 
  -- Compilation date = XX2022-08-08XX
  -- Configuring done
  -- Generating done
  -- Build files have been written to: E:/Project/jni/code4/app/.cxx/Debug/47161615/codec2/src/codec2_native
  [47/202] Performing build step for 'codec2_native'
  ninja: no work to do.
  [48/202] Performing install step for 'codec2_native'
  FAILED: E:/Project/jni/code4/app/.cxx/Debug/47161615/codec2/src/codec2_native-prefix/src/codec2_native-stamp/codec2_native-install 
  cmd.exe /C "cd /D E:\Project\jni\code4\app\.cxx\Debug\47161615\codec2\src\codec2_native && D:\android\sdk\cmake\3.18.1\bin\cmake.exe -E copy E:/Project/jni/code4/app/.cxx/Debug/47161615/codec2/src/codec2_native/src/generate_codebook E:/Project/jni/code4/app/.cxx/Debug/47161615/codec2/src && D:\android\sdk\cmake\3.18.1\bin\cmake.exe -E touch E:/Project/jni/code4/app/.cxx/Debug/47161615/codec2/src/codec2_native-prefix/src/codec2_native-stamp/codec2_native-install"
  Error copying file "E:/Project/jni/code4/app/.cxx/Debug/47161615/codec2/src/codec2_native/src/generate_codebook" to "E:/Project/jni/code4/app/.cxx/Debug/47161615/codec2/src".
  ninja: build stopped: subcommand failed.

How to resolve this error？

tmiw, 2022-08-08:
For cross-compiling, it's recommended to create a toolchain file and pass that into cmake using `-DCMAKE_TOOLCHAIN_FILE`. An example of what such a file could look like when compiling for Windows on Linux or macOS is below:

```
set(CMAKE_SYSTEM_NAME Windows)
set(CMAKE_SYSTEM_PROCESSOR aarch64)

set(triple aarch64-w64-mingw32)

set(CMAKE_C_COMPILER ${triple}-gcc)
set(CMAKE_C_COMPILER_TARGET ${triple})
set(CMAKE_CXX_COMPILER ${triple}-g++)
set(CMAKE_CXX_COMPILER_TARGET ${triple})

set(HOST ${triple})
```

Of course, you'll need to adjust the above to match the locations of the Android compilers and the platform name Android expects. Hope that helps!

drowe67, 2022-08-13:
duplicate

closed 2022-08-13 by drowe67