drowe67/codec2-dev #347: error building on macOS 10.14

ra1nb0w, 2022-08-10:
When building the last commit on macOS 10.14 I get the following error
```
:info:build In file included from /opt/local/var/macports/build/_Users_rainbow_sources_github_macports-ports_audio_codec2/codec2/work/codec2-6f2553ba2b8ffc6f84eba767ebb96
715878477dc/src/fdmdv_demod.c:45:
:info:build /opt/local/var/macports/build/_Users_rainbow_sources_github_macports-ports_audio_codec2/codec2/work/codec2-6f2553ba2b8ffc6f84eba767ebb96715878477dc/src/freedv
_api.h:142:9: error: duplicate member 'interleave_frames'
:info:build     int interleave_frames; // currently ignored, was previously used to configure 700D interleaver
:info:build         ^
:info:build /opt/local/var/macports/build/_Users_rainbow_sources_github_macports-ports_audio_codec2/codec2/work/codec2-6f2553ba2b8ffc6f84eba767ebb96715878477dc/src/freedv
_api.h:140:9: note: previous declaration is here
:info:build     int interleave_frames;                   // now unused but remains to prevent breaking API for legacy apps
:info:build         ^
:info:build 1 error generated.
:info:build make[2]: *** [src/CMakeFiles/fdmdv_demod.dir/fdmdv_demod.c.o] Error 1
:info:build make[2]: Leaving directory `/opt/local/var/macports/build/_Users_rainbow_sources_github_macports-ports_audio_codec2/codec2/work/build'
:info:build make[1]: *** [src/CMakeFiles/fdmdv_demod.dir/all] Error 2
```

tmiw, 2022-08-10:
I'm running 12.4 and am able to compile 6f2553ba (aka latest master) with no problems. Can you confirm what commit you're trying to compile?

ra1nb0w, 2022-08-10:
it is in the log. `6f2553ba2b8ffc6f84eba767ebb96`
tomorrow I re-check building the code out of macports to track down the double declaration

ra1nb0w, 2022-08-11:
Sorry. My fault. I was applying an old patch to fix `interleave_frames`. For some reason it doesn't fail now.
Have a great day and thanks for your attention.

closed 2022-08-11 by ra1nb0w