drowe67/codec2-dev #356: CMake Error

gvanem, 2022-10-10:
While trying to use CMake to configure for MinGW, I got this error:
```
...
CMake Error at CMakeLists.txt:256 (if):
  if given arguments:

    "EQUAL" "8"

  Unknown arguments specified
```

I use CMake version 3.24. But with this patch, cmake runs fine:
```diff
--- a/CMakeLists.txt 2022-10-10 08:23:05
+++ b/CMakeLists.txt 2022-10-10 08:38:37
@@ -253,7 +253,7 @@
     #

     # Detect if we're doing a 32-bit or 64-bit windows build.
-    if(${CMAKE_SIZEOF_VOID_P} EQUAL 8)
+    if("${CMAKE_SIZEOF_VOID_P}" STREQUAL "8")
         set(CMAKE_CL_64 TRUE)
     endif()
```

Comments?

tmiw, 2022-10-10:
FWIW, I use version 3.23.3 on macOS and don't have issues. Perhaps this is a requirement that got introduced in 3.25?

Anyway, thanks for letting us know. 👍  If you're able to submit a pull request, we'd be glad to take a look and merge it up.

gvanem, 2022-10-11:
>  on macOS and don't have issues. 

Surprise. The diff is inside the `WIN32` section. 
But I'm not familiar with CMake, so someone else who are on Windows should test it.

tmiw, 2022-10-11:
> > on macOS and don't have issues.
> 
> Surprise. The diff is inside the `WIN32` section. But I'm not familiar with CMake, so someone else who are on Windows should test it.

Typically, Codec2 for Win32 is built via [the Docker container in freedv-gui](https://github.com/drowe67/freedv-gui/tree/master/docker) and that currently seems to work okay. Then again, the version of CMake in the container may not be the same one you're using.

Can you explain a bit more about your setup? I have a virtualization setup here that I may be able to use to duplicate it.

gvanem, 2022-10-11:
> Can you explain a bit more about your setup?

On Win-10 with a .bat-script invoking this:
```bat
cmake -G "MinGW Makefiles"  -DCMAKE_C_COMPILER=f:/MinGW32/TDM-gcc/bin/gcc.exe
 -DCMAKE_C_FLAGS=-m64 -DCMAKE_MAKE_PROGRAM=make.exe -Wno-dev 
 -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=1 -DCMAKE_SYSTEM_PROCESSOR:STRING=x64 ..
```

drowe67, 2022-10-11:
Yes I think we generally cross compile for Windows on Linux machines, we are mainly concerned with building libcodec2.

@tmiw  we should probably make a PLT level decision around support of native builds on Windows.  I'm not inclined too, seems like we have that covered already via cross compiling and all our development is on Unix type systems.  Cross compiling makes automation much easier too.

tmiw, 2022-10-12:
I did a bit more research on the error in OP and it might not be a problem on our end. The exact error message seems to indicate that `${CMAKE_SIZEOF_VOID_P}` isn't getting set, when in fact CMake is supposed to be doing so. The original proposed workaround was enough to get past the error but I suspect there'd be issues when it came time to generating a Windows installer given the location of the error.

@gvanem, do you see anything in CMakeOutput.log or CMakeError.log around that specific variable or any other general error messages?

gvanem, 2022-10-12:
@tmiw I give on understanding CMake. Seems `MINGW` (which is the only target usable for Windows) is not detected as `WIN32`
by CMake. Go figure.

tmiw, 2022-10-13:
> @tmiw I give on understanding CMake. Seems `MINGW` (which is the only target usable for Windows) is not detected as `WIN32` by CMake. Go figure.

That probably isn't the reason for this particular issue given that the Docker container is able to instantiate `${CMAKE_SIZEOF_VOID_P}` without a problem. Regardless, we'll keep this on hold for now until we get more info (e.g. CMakeOutput.log / CMakeError.log).

drowe67, 2023-02-22:
The PLT agreed not to directly support native Windows builds, README.md has been modified to avoid any future confusion.  Thanks to all for helping us bring this one to light.

closed 2023-02-22 by drowe67