drowe67/codec2-dev #357: Binary stdin/stdout on Windows

gvanem, 2022-10-11:
None of the examples I've tried that pipes data between stdout/stdin and to e.g `sox` works on Windows.
E.g. this command (based on `script\train_700c_quant.sh`):
```
c2enc.exe 700C ./raw/hts1a.raw - --var | c2dec.exe 700C - - | sox -q -t .s16 -c 1 -r 8000 -b 16 - hts1a.wav
./raw/hts1a.raw var:  9.33 std:  3.05
```

generated this `hts1a.wav` that sound like:

https://user-images.githubusercontent.com/945271/195076343-4ba984a8-a4c6-4fa6-82b0-f82b2b276995.mp4

Nice! A `setmode(stdin, O_BINARY)` is needed to fix this. Here's my [attempt](https://github.com/drowe67/codec2/files/9754593/bin-mode-fix.diff.txt) to fix it. So after that patch that same sounds like:

https://user-images.githubusercontent.com/945271/195077897-2ec13f9f-0399-48ba-815b-4d8ef29fec98.mp4

*The Navy attacks the V task force*?

drowe67, 2022-10-11:
As per https://github.com/drowe67/codec2/issues/356#issuecomment-1275161545 - I'm not sure this is an Issue - more a policy that all our development and tool support is on Unix based environments. 

I'm not interested in supporting Windows C/shell incompatibilities, seems like a corner case to me.

@tmiw - we should probably make an official call on this, and document it in the READMEs.

gvanem, 2022-10-12:
> I'm not interested in supporting Windows C/shell incompatibilities, 

It's not a shell issue. 
And he [front-page](https://github.com/drowe67/codec2#building-for-windows-on-a-linux-machine) mentions *Building for Windows ...*, so that gave me the impression Windows is supported. Not the case I guess.

tmiw, 2022-10-12:
I would consider Codec2 supported on Windows (otherwise, the main FreeDV application would have issues running on that platform, too), but I'm not sure if we've attempted to run the ctests or helper applications there much. I took a look at `c2enc`/`c2dec` and it seems we do follow best practice when actual filenames are used (e.g. `fopen` using "b" as part of the mode). The main issue is stdin/stdout as previously mentioned.

Anyway, I'd recommend specifying separate filenames as a workaround. For example:

```
c2enc.exe 700C ./raw/hts1a.raw hts1a_temp.raw --var
c2dec.exe 700C hts1a_temp.raw hts1a_out.raw
sox -q -t .s16 -c 1 -r 8000 -b 16 hts1a_out.raw hts1a.wav
```

@drowe67, IMO I'd actually lean towards fixing this issue considering that we do seem to handle it when filenames are provided and that the fix does seem to be pretty minor (but will need to be applied to stuff like `freedv_tx` too, not just `c2enc` or `c2dec`). Let me see if I can spin up a PR real quick.

tmiw, 2022-10-12:
@gvanem, feel free to try https://github.com/drowe67/codec2/pull/358 and let us know if that works any better for you. 👍

drowe67, 2022-12-02:
> @drowe67, IMO I'd actually lean towards fixing this issue considering that we do seem to handle it when filenames are provided and that the fix does seem to be pretty minor (but will need to be [...]

For me it's a question of scope/resourcing - if we decide to support Windows for _every command line tool_ its a very big job - and someone has to maintain it, and it needs to be tested, and this needs to happen forever.  Some of the tools (like `ch`) need files generated for them by Octave.  So supporting a non Unix-compatible command line/shell is a big commitment, and I'm not sure what the actual benefit will be, or how we would resource it.  It won't help me (quite the opposite) - and I'm the primary codec2 developer.  I don't believe anyone has ever asked for this in the past, so demand is low.

I'd be prepared to consider some small subset, like say c2enc/c2dec - if the testing on Windows can be automated.

So - a good one to discuss at team/PLT level - and make a policy decision, that we document in the README. @gvanem concerns are valid, we just need to decide at what level we support the Windows CLI, and make that clear.

PR #365 (2022-12-26): Clarify Windows support in the README
Resolves #357 by clarifying the current extent of the project's support for Windows and providing alternate instructions using the WSL for development.
closed 2022-12-26 by drowe67