drowe67/codec2-dev #363: Reduce size of codec2 repo

drowe67, 2022-12-02:
109Mbyte is a bit big, most of it seems to be in `.git`, so we might just need to rm some branches.

```
david@xps15:~/tmp/codec2$  master $ du -h
944K	./src/codebook
6.3M	./srcavid@xps15:~/tmp/codec2$  master $ du -h
944K	./src/codebook
6.3M	./src
1.5M	./doc
8.0K	./.git/info
8.0K	./.git/refs/heads
8.0K	./.git/refs/remotes/origin
12K	./.git/refs/remotes
4.0K	./.git/refs/tags
28K	./.git/refs
4.0K	./.git/branches
8.0K	./.git/logs/refs/heads
8.0K	./.git/logs/refs/remotes/origin
12K	./.git/logs/refs/remotes
24K	./.git/logs/refs
32K	./.git/logs
4.0K	./.git/objects/info
84M	./.git/objects/pack
84M	./.git/objects
56K	./.git/hooks
84M	./.git
6.5M	./raw
48K	./script
4.1M	./octave
4.2M	./wav
44K	./demo
436K	./unittest
12K	./.github/workflows
16K	./.github
136K	./misc
232K	./stm32/src
704K	./stm32/doc
64K	./stm32/inc
56K	./stm32/usb_conf
28K	./stm32/stlink
164K	./stm32/unittest/src
8.0K	./stm32/unittest/lib/octave
8.0K	./stm32/unittest/lib/python
24K	./stm32/unittest/lib
152K	./stm32/unittest/scripts
356K	./stm32/unittest
152K	./stm32/usb_lib/otg
68K	./stm32/usb_lib/core
48K	./stm32/usb_lib/cdc
272K	./stm32/usb_lib
40K	./stm32/cmake
1.8M	./stm32
16K	./cmake
109M	.
```

tmiw, 2022-12-03:
Seems to be 82MB for me:

```
mooneer@ubuntu:~$ git clone git@github.com:drowe67/codec2.git
Cloning into 'codec2'...
X11 forwarding request failed on channel 0
remote: Enumerating objects: 26641, done.
remote: Counting objects: 100% (471/471), done.
remote: Compressing objects: 100% (182/182), done.
remote: Total 26641 (delta 300), reused 450 (delta 289), pack-reused 26170
Receiving objects: 100% (26641/26641), 82.47 MiB | 8.59 MiB/s, done.
Resolving deltas: 100% (19642/19642), done.
mooneer@ubuntu:~$
```

Also by branch (per [Stack Overflow](https://stackoverflow.com/questions/32557849/get-git-branch-size)), though this doesn't seem to be catching everything:

```
mooneer@ubuntu:~/codec2$ git for-each-ref --format='%(refname)' |
while read branch
do
    size=$(git rev-list --disk-usage --objects HEAD..$branch)
    echo "$size $branch"
done |
sort -n
0 refs/heads/master
0 refs/remotes/origin/dr-lpcnet-circ
0 refs/remotes/origin/HEAD
0 refs/remotes/origin/master
0 refs/remotes/origin/ms-embedded-noarm
0 refs/remotes/origin/ms-esp-idf-failure
0 refs/remotes/origin/ms-macos-older-releases
0 refs/remotes/origin/ms-test-cmake
0 refs/tags/stm32_PR#317
0 refs/tags/v0.9.2
0 refs/tags/v1.0.1
0 refs/tags/v1.0.3
0 refs/tags/v1.0.4
0 refs/tags/v1.0.4_rc2
0 refs/tags/v1.0.5
132 refs/tags/v1.0.0
11835 refs/remotes/origin/dr-data
81323 refs/remotes/origin/ms-win32-binary
708773 refs/remotes/origin/dr-papr
mooneer@ubuntu:~/codec2$
```

drowe67, 2022-12-03:
> Seems to be 82MB for me:

Yes I had the same size transfer on the `git clone`, compression I guess.

Artoria2e5, 2023-06-12:
> though this doesn't seem to be catching everything:
Thing seems to compare branch vs HEAD, so whatever history is in the HEAD won't show up. That does confirm the master branch is the big one with loads of history. I *guess* the thing to do is a shallow clone for now, because a giant flattening is going to be very bad on existing PRs.

The fun recipe from https://stackoverflow.com/a/46085465 gives something more useful, in case we end up doing some filter-branch or flattening in the future:
```
$ git rev-list --objects --all | git cat-file --batch-check='%(objecttype) %(objectname) %(objectsize) %(rest)' | sed -n 's/^blob //p' | sort --numeric-sort --key=2 | tail -n 10 | cut -c 1-12,41- | $(command -v gnumfmt || echo numfmt) --field=2 --to=iec-i --suffix=B --padding=7 --round=nearest
38454a58eb2a  1.7MiB wav/ve9qrp.wav
6457d109d096  1.9MiB octave/tofdm_out.txt
e393d0197b98  1.9MiB octave/tfdmdv_out.txt
d31a1484f630  2.8MiB unittest/noise_samples.h
fc9a9c786e16  3.0MiB unittest/noise_samples.h
107ad9a95c21  7.3MiB raw/fast_fading_samples.float
efc6fa7ac4b4  7.3MiB raw/faster_fading_samples.float
2d22fd5dd032   11MiB octave/tcohpsk_out.txt
56478683506a   34MiB raw/ve9qrp_multi.raw
2f7f22244ab8   37MiB octave/octave-workspace
```

Note the SHA1 hashes are for the objects (i.e. files), not the commit that includes them.

> 708773 refs/remotes/origin/dr-papr

Not that big considering the size of the PR! Compression in action.

drowe67, 2023-06-14:
Thanks @Artoria2e5 - that's very useful.  I'm planning a general clean up of codec2 soon

drowe67, 2023-07-07:
A very shallow clone (just a new repo from the zip files) is in [codec2-new](https://github.com/drowe67/codec2-new), about 13Mbyte