drowe67/codec2-dev #370: Frequency readout in 800xa is reversed

DG9BFC, 2022-12-27:
title says it .. the frequency redout in 800xa is wrong
when i worked fullduplex over qo100 and tested the different modes i have found out that the 800xa shows the frequency wrong
as a test i set my TX 180Hz HIGHER ... mode 1600 shows PLUS 180 offset while 800xa shows MINUS 180Hz
so in all modes the offset is shown but in 800xa (there the "fault" is shown)
in the two screenshots you see that the tx frequency is higher (or the rx is LOW)
i used usb in both cases and as can be seen signal is shifted to right (higher tones)
800xa shows fault (-180) and other modes show where to tune (+180) to get to 0Hz offset ... one is wrong ;-)
[image]
[image]

drowe67, 2022-12-27:
Thanks @DG9BFC - I think I've fixed it.  It's also interesting to know the use cases for FreeDV, I was wondering if 1600 was in use.

@tmiw - it might be useful to compile a list of use cases for FreeDV, e.g. what modes are used where.  Useful input for development and maintenance.

DG9BFC, 2022-12-27:
hi david .. welcome :-)
i would like to have the levels between the modes more even  but thats a different thingy .. a new release to try ??
about modes ... qo100 is a linear transponder so as long as your station is stable on receive and transmit any freedv mode does work
and cause we all use fullduplex (or should use) you can hear what your qso partner hears (easy to test all modes without a partner or without a websdr involved)
the lnb may add some frequency varation (slight wobble) so modes where a pilot tone is added or were the mode itself is immune to those slight varations work best ... if you need some recordings (or screenshots like above) ... just ask

DG9BFC, 2022-12-27:
sidenote ... with 800xa i can easier drive my station to higher power and get the highest snr reports over the bird (about 20db max) while other modes are more in the 10 to 12 db range

drowe67, 2022-12-27:
Yes 800XA is FSK, so PAPR=0dB, you get the most out of your PA.

DG9BFC, 2022-12-27:
i wait for mooneer for a next test version ... then do further testing
not only that i CAN drive it higher but it IS higher from the soft towards sdr soft
[image]
[image]
here with reduced power
[image]
[image]
and here with higher output

DG9BFC, 2022-12-27:
as can be seen 800xa is much higher

DG9BFC, 2022-12-27:
a 10 db higher power on the 1600 (and other modes) would make the power more even (so comparing modes would be easier)

tmiw, 2022-12-28:
> @tmiw - it might be useful to compile a list of use cases for FreeDV, e.g. what modes are used where. Useful input for development and maintenance. Perhaps something we can discuss at PLT, give the range of experience of the team.

FWIW, 800XA and 1600 seem to be mostly used on linear satellites these days while 700D and E seem to be the most common day to day modes on HF.

drowe67, 2022-12-28:
I guess it's the ability to handle large and rapidly changing frequency offsets.

tmiw, 2022-12-28:
> I guess it's the ability to handle large and rapidly changing frequency offsets.

At least in the case of 1600, the center carrier is much easier to track during a satellite pass for Doppler adjustment purposes. :)

drowe67, 2023-02-22:
@tmiw - Would you agree this one is closed?

tmiw, 2023-02-22:
> @tmiw - Would you agree we can close this one?

Yep 👍

DG9BFC, 2023-02-22:
yes ... readout is fixed
thank you 
dg9bfc sigi

closed 2023-02-22 by DG9BFC