drowe67/codec2-dev #383: Not able to use `all` or `install` targets when cross-compiling

tyler-gilbert, 2023-03-04:
I get linker errors when I try to execute the `all` or `install` targets when building using:

```
PATH=<path to arm none eabi compiler>
cmake \
-DCMAKE_SYSTEM_NAME=Generic \
-DCMAKE_SYSTEM_PROCESSOR=arm \
-DCMAKE_C_COMPILER=arm-none-eabi-gcc \
-DCMAKE_CXX_COMPILER=arm-none-eabi-g++ \
-DCMAKE_C_FLAGS='-mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -Os' \
-DCMAKE_EXE_LINKER_FLAGS='--specs=nosys.specs' \
-DCMAKE_INSTALL_PREFIX=<my output directory> \
-DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=NEVER \
-DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY \
-DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY \
-DCMAKE_FIND_ROOT_PATH_MODE_PACKAGE=ONLY \
-DCMAKE_CROSSCOMPILING=ON \
-DBUILD_SHARED_LIBS=OFF \
-GNinja \
..

With this setup: `ninja codec2` works bug `ninja install` fails with linker errors.

I will send a PR that adds a CMAKE build option called `BUILD_EXECUTABLES` which will exclude executables from the build so that `ninja install` works with embedded.

drowe67, 2023-03-04:
Not sure this is an actual issue - we build for the stm32  (see sm1000) already.  I'm not familiar with ninja, so we need to decide if we will support it going forward. Any thoughts @tmiw ?

tyler-gilbert, 2023-03-04:
This also applies to `make all` or `make install` as well when using the Makefiles generator.

tmiw, 2023-03-04:
What's the use case, exactly? Typically for embedded targets codec2 is statically linked into the .bin or .hex file that ultimately gets flashed.

Also, cross-compiling with CMake typically works by setting the needed compiler, etc. arguments into another file and specifying that file with `-DCMAKE_TOOLCHAIN_FILE=...`. More information can be found [here](https://cmake.org/cmake/help/latest/manual/cmake-toolchains.7.html).

tyler-gilbert, 2023-03-04:
The use case is that I have several third party dependencies (or which codec2 is one of them) that I want to install to a sysroot for an arm-noni-eabi target. I use `install` so that all the headers and static libs for the various dependencies end up in a common sysroot.

The toolchain file has the same effect as what I am passing on the command line. I am able to have everything cross compile. It creates `libcodec2.a` in `<build-folder>/src/`. I want to use `install` to copy that file to the sysroot. But `install` implies `all`. And `all` includes the executables that won't build. So it would be nice to have an option of only building and install the lib and not the executables.

tmiw, 2023-03-04:
I think it would be good to mention what linker errors you're getting since the proposed change may just mask the actual issue. In a sysroot type setting, for instance, I can see some still wanting the tools (which apparently don't build properly for some reason).

tyler-gilbert, 2023-03-04:
I am getting undefined errors in the linking process. For example:

```
  /Users/tgil/git-rs/ptt-rs/deps/src/arm-gnu-toolchain-12.2.rel1-darwin-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Users/tgil/git-rs/ptt-rs/deps/src/arm-gnu-toolchain-12.2.rel1-darwin-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg.a(libc_a-readr.o): in function `_read_r'
```

Ultimately, I want to run this on a system that does not use any system calls. 

With this PR, the default behavior is to build the executables. It just gives the option to skip that step. I think there is a value add either way because you also might want to skip building the executables for expediency.

tmiw, 2023-03-05:
I'm not sure we necessarily want to make the executables work on systems without library calls, but we'll think on this some more.

drowe67, 2023-03-17:
I'm closing this an an issue as it's not around a currently supported feature, it's more feature request for a use case we have not considered and do not support at this time.  For new features or project polices suggestions please feel free to start a Discussions thread.

If there is significant demand or support for the feature we can consider it at out monthly team meetings.

closed 2023-03-17 by drowe67