drowe67/codec2-dev #391: Error compiling for ESP32

happytm, 2023-03-17:
I was trying to compile Codec2 library for ESP32 but I get following error messages:

```Using previously compiled file: C:\Users\WinCycle\AppData\Local\Temp\arduino_build_922383\libraries\codec2\varicode.c.o
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'print_pred_error':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:251:39: error: 'N' undeclared (first use in this function)
      float pred = pexp->phi_prev[i] + N*i*(model->Wo + pexp->Wo_prev)/2.0;
                                       ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:251:39: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'predict_phases':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:266:38: error: 'N' undeclared (first use in this function)
  model->phi[i] = pexp->phi_prev[i] + N*i*model->Wo;
                                      ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'predict_phases_state':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:314:42: error: 'N' undeclared (first use in this function)
      model->phi[i] = pexp->phi_prev[i] + N*i*best_Wo;
                                          ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'predict_phases2':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:333:29: error: 'N' undeclared (first use in this function)
  pred = pexp->phi_prev[i] + N*i*model->Wo;
                             ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'limit_prediction_error':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:561:29: error: 'N' undeclared (first use in this function)
  pred = pexp->phi_prev[i] + N*i*(model->Wo + pexp->Wo_prev)/2.0;
                             ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'quant_prediction_error':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:580:29: error: 'N' undeclared (first use in this function)
  pred = pexp->phi_prev[i] + N*i*(model->Wo + pexp->Wo_prev)/2.0;
                             ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'print_sparse_pred_error':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:607:33: error: 'N' undeclared (first use in this function)
      pred = pexp->phi_prev[i] + N*i*(model->Wo + pexp->Wo_prev)/2.0;
                                 ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'refine_Wo':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:759:33: error: 'N' undeclared (first use in this function)
      pred = pexp->phi_prev[i] + N*i*Wo;
                                 ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'sparse_vq_pred_error':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:819:29: error: 'N' undeclared (first use in this function)
  pred = pexp->phi_prev[i] + N*i*best_Wo;
                             ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'predict_phases1':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:861:38: error: 'N' undeclared (first use in this function)
  model->phi[i] = pexp->phi_prev[i] + N*i*best_Wo;
                                      ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'smooth_phase':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:893:29: error: 'N' undeclared (first use in this function)
  pred = pexp->phi_prev[m] + N*m*best_Wo;
                             ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'smooth_phase2':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:995:30: error: 'N' undeclared (first use in this function)
   pred = pexp->phi_prev[i] + N*i*best_Wo;
                              ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'smooth_phase3':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:1042:18: error: 'FS' undeclared (first use in this function)
  f = m*model->Wo*FS/TWO_PI;
                  ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:1054:33: error: 'N' undeclared (first use in this function)
      pred = pexp->phi_prev[m] + N*m*best_Wo;
                                 ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'cb_phase1':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:1113:18: error: 'FS' undeclared (first use in this function)
  f = m*model->Wo*FS/TWO_PI;
                  ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:1145:39: error: 'N' undeclared (first use in this function)
   model->phi[m] = pexp->phi_prev[m] + N*m*best_Wo;
                                       ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'cb_phase2':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:1169:67: error: 'N' undeclared (first use in this function)
     model->phi[1] = pexp->phi_prev[1] + (pexp->Wo_prev+model->Wo)*N/2.0;
                                                                   ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'repeat_phases':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:1226:19: error: 'N' undeclared (first use in this function)
  model->phi[m] += N*m*model->Wo;
                   ^
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c: In function 'phase_experiment':
C:\Users\WinCycle\Documents\Arduino\libraries\codec2\phaseexp.c:1350:19: error: 'N' undeclared (first use in this function)
     pexp->phi1 += N*model->Wo;
                   ^
Using library codec2 in folder: C:\Users\WinCycle\Documents\Arduino\libraries\codec2 (legacy)
exit status 1
Error compiling for board ESP32 Dev Module.
```
Thanks.

happytm, 2023-03-17:
Following is a code I was using:

```#include <codec2.h>

#define AUDIO_SAMPLE_RATE     8000  // 44100

TaskHandle_t audio_task_;
struct CODEC2* c2_;
int c2_samples_per_frame_;
int c2_bytes_per_frame_;
int16_t *c2_samples_;
uint8_t *c2_bits_;

void setup() {
  
  // run codec2 audio loopback on a separate task
  xTaskCreate(&audio_task, "audio_task", 32000, NULL, 5, &audio_task_);
}

void audio_task(void *param) {
  // construct and configure codec2
  c2_ = codec2_create(CODEC2_MODE_450);
  if (c2_ == NULL) {
  
    return;
  } else {
    //codec2_set_lpc_post_filter(c2_, 1, 0, 0.8, 0.2);
    c2_samples_per_frame_ = codec2_samples_per_frame(c2_);
    c2_bytes_per_frame_ = codec2_bytes_per_frame(c2_);
    c2_samples_ = (int16_t*)malloc(sizeof(int16_t) * c2_samples_per_frame_);
    c2_bits_ = (uint8_t*)malloc(sizeof(uint8_t) * c2_bytes_per_frame_);
   
  }

  // run loopback record-encode-decode-playback loop
  size_t bytes_read, bytes_written;
  
  while(true) {
    codec2_encode(c2_, c2_bits_, c2_samples_);
    vTaskDelay(1);
    codec2_decode(c2_, c2_samples_, c2_bits_);
     vTaskDelay(1);
  }
}

void loop() {
  // do nothing
  delay(100);
}
```

drowe67, 2023-03-17:
I'd like to keep Issues for problems with Codec 2 on it's target operating systems and platforms.  This does not (officially at least) include Windows, as we lack the resources to support this. 

Can you please move this thread to Discussions?

closed 2023-03-17 by drowe67