drowe67/codec2-dev #394: Problem with similar functions of different codecs

Jamaika1, 2023-04-01:
Problem with similar functions of different codecs in ffmpeg compilation.
I managed to compile codec2 in ffmpeg. However I had trouble with similar features in other codecs.
https://github.com/Jamaika1/codec2/commit/218c3857a2db97b784ae9e316aa78db2e1e9e1ed#diff-a26553a2385e9f1f6e0d85e2ef7965602fc8f8049de23c60bad5ff3f57ebfe71

https://github.com/Jamaika1/plugins_ffmpeg_vvc_evc_htj2k/tree/main/libavcodec

tmiw, 2023-04-01:
For the commit you linked to, are these the changes you had to make to get codec2 working, or to fix the issues in the other codecs? What issues did you have? It's not immediately clear from the links provided.

Jamaika1, 2023-04-01:
Hi tmiw
I had  problem with common functions e.g. `lsp_to_lpc` in speex.
https://github.com/xiph/speex

tmiw, 2023-04-02:
It would be good to know how exactly you're trying to build Codec2 into ffmpeg as I'm not sure it's necessary to make the changes that you've made. If you can provide that info, we can take a look and provide suggestions.

tmiw, 2023-04-04:
Closing as building codec2 is not officially supported on Windows at this time. See https://github.com/drowe67/codec2/pull/365 for details.

That said, I suspect it's some kind of build environment issue. For instance, the ffmpeg Windows binaries at https://gyan.dev/ffmpeg/builds also include codec2 and presumably didn't have to change anything code-wise to make it happen. It may be worthwhile to take a look at what they did and integrate their changes into your ffmpeg build as appropriate.

closed 2023-04-04 by tmiw