drowe67/codec2-dev #395: Android project implementation

shlomi-commcrete, 2023-04-03:
Hi, Im having trouble importing the codec2 library into my android app correctly, 

Could anyone help me? 

I could not bring it to work using the instructions here.

tmiw, 2023-04-03:
What issue are you having? Maybe the comments [here](https://github.com/drowe67/codec2/issues/346) may help?

shlomi-commcrete, 2023-04-03:
The issue im having is the implimentation.

Iv'e been following this instructions : 

Including Codec 2 in an Android project
In an Android Studio 'NDK' project (a project that uses 'native' code) Codec 2 can be added to the project in the following way.

Add the Codec 2 source tree to your app (e.g. in app/src/main/codec2) (e.g. as a git sub-module).

Add Codec 2 to the CMakeList.txt (app/src/main/cpp/CMakeLists.txt):

# Sets lib_src_DIR to the path of the target CMake project.
set( codec2_src_DIR ../codec2/ )
# Sets lib_build_DIR to the path of the desired output directory.
set( codec2_build_DIR ../codec2/ )
file(MAKE_DIRECTORY ${codec2_build_DIR})

add_subdirectory( ${codec2_src_DIR} ${codec2_build_DIR} )

include_directories(
    ${codec2_src_DIR}/src
    ${CMAKE_CURRENT_BINARY_DIR}/../codec2
)
Add Codec 2 to the target_link_libraries in the same file.

So what i understand is this  : 
1. i need to add a sub module containing a CMakeLists.txt
2. what does : Add Codec 2 to the target_link_libraries in the same file.
does that mean calling the library from my settings.gradle/build.gradle (implementation project(":codec2"))?
3. How do i call the C code from my native app?

[image]
[image]

tmiw, 2023-04-03:
I haven't done much Android development lately (and what I've done has been in Java), so I can't comment on question 3. However, for question 2, the file it's referring to is the CMakeLists.txt file. For example:

```
target_link_libraries(your_binary_name codec2 [whatever other libraries your application needs])
```