drowe67/codec2-dev #397: FSK demodulator regression?

scd31, 2023-04-30:
I've been doing some work updating [Wenet](https://github.com/projecthorus/wenet) to use the current version of fsk_demod.c.  For context, Wenet ships with an fsk_demod.c from 2019. It works fine for 2-FSK and I can confirm that I can use one of the 2019 commits from this repository with it, without issue. Also note that I'm running at 500kbps with 4x oversampling.

Using the latest fsk_demod, however, introduces a problem. No longer do all packets decode successfully, even though the SNR is quite high. Experimentally, I've found that setting the option `--nsym 300` helps a lot, though I still drop more packets than I should. Every image tends to have at least a few missing packets - with the old decoder, a missing packet is extremely rare, which makes sense given the high SNR.

I tried to do a git bisect but over such a large range it's not very helpful. It also looks like at one point fsk_demod didn't work at all - I'd get no decoded frames out - and this made the bisect quite useless.

I have more information in an issue I opened in that repository: https://github.com/projecthorus/wenet/issues/5

Tyrbiter, 2023-05-04:
You might have a look at this, from 2020, and see if anything described there looks like it could change the performance to what you're seeing.

https://github.com/drowe67/codec2/commit/96414249f09b9b20c2588d54bc4d34a343a6c004

I'm in no way an expert on this, there are roughly 20 commits to the fsk_demod.c file since 2018, so you could look at the history to see what they are. I don't know if other source code changes might affect the code in this file.

scd31, 2023-05-04:
Unfortunately I don't think that would be the cause as I'm setting `p` manually. I'm currently squashing bugs unrelated to codec2 so this is on the back burner for now but I hope to get it sorted out in the coming weeks. Thanks!

drowe67, 2023-05-04:
I'm going to close this as (a) its already an issue on Project Horus and (b) it's more an integration issue with a fork of codec2 than a demonstrable bug in the master branch of `drowe67/codec2`.  Happy to continue the discussion over on Project Horus or reopen here if it can be shown to be an actual issue in the master branch of `drowe67/codec2`.