drowe67/codec2-dev #399: the robustness to the single bit error?

moseranliunian, 2023-05-14:
Sorry to bother you, but I would like to ask one more question about codebook sorting. It is embodied in what method is adopted to achieve the robustness to the single bit error  in 450bps ? And in what programming files is the calculation of the quality factor Q represented? Or is there a picture?

drowe67, 2023-05-20:
Could you please post these questions to Discussions?  They are not actually Issues with Codec 2 that need fixing.

This blog post https://www.rowetel.com/?p=7837 has references to papers on index sorting.

closed 2023-05-20 by drowe67

moseranliunian, 2023-05-22:
Oh, I'm really sorry. Since this is the first time for me to learn about voice codec and GitHub, I am not very familiar with some relevant regulations