drowe67/codec2-dev #409: Problem with tiny speakers

programmer131, 2024-04-15:
Many of the tiny sized speakers have frequency range above 1KHz, but most of the decoded voice signal components are below 1KHz, do you have any suggestions or design recommendations for interfacing small speakers? do codec2 has built-in pitch shifting algorithm that I can tune? I've tried to play high frequency signal (in 4 to 8KHz range) and speaker produce high volume sound but same speaker suppress most of the audio that is decoded from codec2, because of low frequency components.

tmiw, 2024-04-15:
If you use [the FreeDV application](https://github.com/drowe67/freedv-gui) there are some filter settings that you can adjust there to bump up the lower frequencies when transmitting and receiving. I don't recall offhand if there's anything in Codec2 itself, though.

programmer131, 2024-04-16:
this is esp32 to esp32 voice communication using codec2, not sure how to extract that frequency shift algo from FreeDV.

tmiw, 2024-04-16:
> this is esp32s3 to esp32s3 voice communication using codec2, not sure how to extract that frequency shift algo from FreeDV.

The filter stuff in FreeDV uses sox to implement it so you'd have to integrate that library (or equivalent).

There's also `codec2_set_lpc_post_filter()` in the codec2 library, but that's only good for about a 3 dB boost. Example:

```c
        struct CODEC2 *c2 = freedv_get_codec2(dv);
        if (c2 != NULL) 
        {
            codec2_set_lpc_post_filter(c2, enable, bassBoost, beta, gamma);
        }
```

(The defaults for `beta` and `gamma` are 0.20 and 0.50 respectively. To turn on the bass boost, you'd need to set `enable` and `bassBoost` to `true`.)

drowe67, 2024-04-17:
@programmer131 - Codec 2 is used in many use cases with small speakers (e.g. HF and VHF hand held radios), but these tend to have a freq response that extends down to 400 Hz or so. Sorry, we do not have any specific features for your use case.

closed 2024-04-17 by drowe67